openhwfoundation/cv32e40x #20: VCS Compiler error building cv32e40x

treforsouthwell, 2021-03-29:
I tried to build the cv32e40x_core as checked into Git today (no modifications) and hit a compile error:

Error-[TFIPC-L] Too few instance port connections
/user/trs/work/trefor.southwell_SWANN/volcanic/MAIN/design/gen_PULAR_CONFIG21_VCS_sim_rgx_rogue4_tb/import/cv32e40x/rtl/cv32e40x_core.sv, 359
cv32e40x_core, "cv32e40x_id_stage #(.USE_PMP(USE_PMP), .A_EXTENSION(A_EXTENSION)) id_stage_i( .clk (clk),  .clk_ungated_i (clk_i),  .rst_n (rst_ni),  .scan_cg_en_i (scan_cg_en_i),  .fetch_enable_i (fetch_enable),  .ctrl_busy_o (ctrl_busy),  .is_decoding_o (is_decoding),  .instr_req_o (instr_req_int),  .branch_decision_i (branch_decision),  .jmp_target_o (jump_target_id),  .clear_instr_valid_o (clear_instr_valid),  .pc_set_o (pc_set),  .pc_mux_o (pc_mux_id),  .exc_pc_mux_o (exc_pc_mux_id),  .exc_cause_o (exc_cause),  .halt_if_o (halt_if),  .id_ready_o (id_ready),  .ex_ready_i (ex_ready),  .wb_ready_i (lsu_ready_wb),  .id_valid_o (id_valid),  .ex_valid_i (ex_valid),  .if_id_pipe_i (if_id_pipe),  .id_ex_pipe_o (id_ex_pipe),  .current_priv_lvl_i (current_priv_lvl),  .csr_c ... "
  The above instance has fewer port connections than the module definition,
  There are 70 port(s) in module "cv32e40x_id_stage" definition, but only 68
  port connect(s) in the instance.
  input port 'data_err_i' is not connected,
  output port 'data_err_ack_o' is not connected.

Note - this is compiled outside the scripts in GIT using an alternative EDA flow.

Silabs-ArjanB, 2021-03-30:
Hi @treforsouthwell Thank you for filing this issue. This related to work in progress related to adding support for bus errors. For our tools this leads to a warning. The pin lists should be fixed now (but the bus error feature is not present yet).

Silabs-ArjanB, 2021-03-31:
Hi @treforsouthwell Could you please close this issue if you agree it has been fixed?

treforsouthwell, 2021-03-31:
Thanks for the fix, I'm afraid it seems someone has (in parallel) removed the port and now we get a build error for the port not existing:

Top Level Modules:
       cv32e40x_core
TimeScale is 1 ns / 1 ps
Final timescale is '1ps/1ps'

Error-[UPIMI-E] Undefined port in module instantiation
/user/trs/work/trefor.southwell_SWANN/volcanic/MAIN/design/gen_PULAR_CONFIG21_DC_syn/import/cv32e40x/rtl/cv32e40x_core.sv, 365
  Port "data_err_i" is not defined in module 'cv32e40x_id_stage' defined in
  "/user/trs/work/trefor.southwell_SWANN/volcanic/MAIN/design/gen_PULAR_CONFIG21_DC_syn/import/cv32e40x/rtl/cv32e40x_id_stage.sv",
  32
  Module instance: cv32e40x_id_stage #(.USE_PMP(USE_PMP),
  .A_EXTENSION(A_EXTENSION)) id_stage_i( .clk (clk),  .clk_ungated_i (clk_i),
  .rst_n (rst_ni),  .scan_cg_en_i (scan_cg_e ...

Error-[UPIMI-E] Undefined port in module instantiation
/user/trs/work/trefor.southwell_SWANN/volcanic/MAIN/design/gen_PULAR_CONFIG21_DC_syn/import/cv32e40x/rtl/cv32e40x_core.sv, 365
  Port "data_err_ack_o" is not defined in module 'cv32e40x_id_stage' defined
  in
  "/user/trs/work/trefor.southwell_SWANN/volcanic/MAIN/design/gen_PULAR_CONFIG21_DC_syn/import/cv32e40x/rtl/cv32e40x_id_stage.sv",
  32
  Module instance: cv32e40x_id_stage #(.USE_PMP(USE_PMP),
  .A_EXTENSION(A_EXTENSION)) id_stage_i( .clk (clk),  .clk_ungated_i (clk_i),
  .rst_n (rst_ni),  .scan_cg_en_i (scan_cg_e ...

1 warning
2 errors
CPU time: .280 seconds to compile

Silabs-ArjanB, 2021-03-31:
@treforsouthwell Apologies for the unfortunate merge that happened between the two latest pull requests. I will fix it right away and will give an update here when done.

Silabs-ArjanB, 2021-03-31:
@treforsouthwell Now the 'core' and its 'id stage' should have matching pins. In core-v-verif we still get some (also related) warnings for the error pin hookup of cv32e40x_wrapper_i. Hopefully that does not impact what you try to do. (I will ask somebody else to have a look at those warnings (the related testbench functionality is not there yet).

treforsouthwell, 2021-03-31:
Thanks, I saw the fix. I'm just running synthesis and was able to workaround this with manual edits but wanted to report the problem.

Silabs-ArjanB, 2021-03-31:
I filed a separate ticket for the core-v-verif fixes, see https://github.com/openhwgroup/core-v-verif/issues/544