openhwfoundation/cv32e40x #30: PMA: Non-aligned transactions are not supported in I/O regions.

eroom1966, 2021-04-20:
In the documentation for the PMA, it says

> Non-aligned transactions are not supported in I/O regions.

What action is taken if a non aligned Load/Store occurs ?
In the exceptions I do not see support for 
Code(4) - Load Address Misaligned, or
Code(6) - Store/AMO address misaligned

Thx
Lee

Silabs-ArjanB, 2021-04-20:
I believe code 5 and 7 are according to the RISC-V privileged spec. We will have the following PMA related exceptions:

- An attempt to perform a non-naturally aligned load access to I/O causes a precise load access fault (exception code 5).
- An attempt to perform a non-naturally aligned store access to I/O causes a precise store access fault (exception code 7).
- An attempt to execute a an instruction from I/O causes a precise instruction access fault (exception code 1).
- An attempt to perform a Load-Reserved to a region in which Atomic operations are not allowed causes a precise load access fault (exception code 5).
- An attempt to perform a Store-Conditional or Atomic Memory Operation (AMO) to a region in which Atomic operations are not allowed causes a precise store/AMO access fault (exception code 7).

I am actually in the process of updating the User Manuals of the 40X and 40S with exactly this type of information. I will release it this week.

Silabs-ArjanB, 2021-04-21:
Hi @eroom1966,

The PMA, exception, and interrupt related descriptions have been updated, specifically see:

- https://cv32e40x-user-manual.readthedocs.io/en/latest/pma.html
- https://cv32e40x-user-manual.readthedocs.io/en/latest/exceptions_interrupts.html

(Similar updates have been made for the CV32E40S User Manual as well).

Please let me know if this addresses your question and if this issue can now be closed.

eroom1966, 2021-04-21:
@Silabs-ArjanB 
Perfect!
Thanks

closed 2021-04-21 by eroom1966