openhwfoundation/cv32e40x #36: PMA Section of user manual missing information

silabs-hfegran, 2021-04-27:
From what I can see, the user manual lacks a description of the following:
- Valid number of PMA regions
- How non-matching regions should be treated

Additionally, I'm wondering if it would be a good idea to define a value for the atomic attribute in case of a lacking A-extension to facilitate assertion re-use. In other words, always tied high for main memory, as rv-spec requires main memory to support whichever atomic operations the core itself supports.

Silabs-ArjanB, 2021-04-27:
With respect to the atomic attribute for cores without A extension, I would prefer if we remove the attribute from the RTL all together.

Silabs-ArjanB, 2021-05-07:
@silabs-hfegran Could you please check if your questions have been addressed in the current User Manual and if so close this issue?

silabs-hfegran, 2021-05-07:
Looks good!

closed 2021-05-07 by silabs-hfegran