openhwfoundation/cv32e40x #39: Negative parameter indexes

MikeOpenHWGroup, 2021-04-28:
A recent addition to the CV32E40X (sha 4206fff4 on 2021-04-27) introduced "negative indexes" for the `PMA_CFG` parameter.  An example of this is `bhv/cv32e40x_wrapper.sv`:
```
module cv32e40x_wrapper
  import cv32e40x_pkg::*;
#(
  parameter NUM_MHPMCOUNTERS             =  1,
  parameter int unsigned PMA_NUM_REGIONS =  0,
  parameter pma_region_t PMA_CFG [PMA_NUM_REGIONS-1:0] = '{default:PMA_R_DEFAULT}
)
```
The default value of `PMA_NUM_REGIONS` is zero which means the range of `PMA_CFG` is [-1:0].  Most SV simulators will accept this, but Metrics DSIM interprets SV parameters as unsigned values (which, I believe is correct) and therefore the size of PMA_CFG is (2**31)-1 which exceeds a built-in DSIM limit.

Silabs-ArjanB, 2021-04-28:
Hi @silabs-oivind I believe my original proposal would prevent this.

silabs-oivind, 2021-04-29:
Hi @MikeOpenHWGroup,
I've updated the CV32E40X RTL to work around this.
However, if I read the section about range specification in the SystemVerilog LRM correctly, range boundaries are allowed to be any integer value, "positive, negative, or zero".

MikeOpenHWGroup, 2021-04-29:
Thanks @silabs-oivind, that did the trick.   Just for safety's sake it would also be good to do the same for the 40x_wrapper.

FWIW, I've looked into this and agree with you that the LRM supports negative range boundaries.  The Metrics DSIM simulator supports these as expected, so I was in error thinking this was the issue.   It turns out the problem is that DSIM interprets parameters as unsigned values, so if `PUM_NUM_REGIONS` is zero, then  `PMA_NUM_REGIONS-1` is resolved as a very large positive number, not -1.

closed 2021-04-29 by MikeOpenHWGroup