openhwfoundation/cv32e40x #47: Use of SystemVerilog Interfaces in RTL

MikeOpenHWGroup, 2021-05-04:
The e40x RTL defines at least one SystemVerilog Interface, `if_c_obi` which is used in several modules.  It is my understanding that OpenHW has standardized on the [Ibex Style Guide](https://github.com/lowRISC/style-guides) for RTL and this guide lists SV interfaces as [problematic](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#problematic-language-features-and-constructs) and their use is discouraged.

The Ibex team does not provided a rationale for that decision, and I have a [pending question](https://github.com/lowRISC/style-guides/issues/52) about it.

Is there a reason that we are using SV Interfaces in the RTL?  I notice that packed structs are used in other areas of the RTL and perhaps it makes sense to standardize that.

Silabs-ArjanB, 2021-05-04:
Hi @MikeOpenHWGroup Although the Ibex style guide is mentioned on our github pages the CV32E40P is not following it nor will the CV32E40X/S follow it. Reason to leave it there is just in case other people contribute, in which case some style guide is better than none. (It is our ambition to make our own style guide based on the Ibex one; some guidlines will simply be removed, some will be changed.)

The reason we are heavily using structs and interfaces is that we are trying to maximize the RTL code re-use between the 40X and the 40S. We have analyzed the expected differences and we are trying to ensure that as many lines of code can remain unchanged between these cores when features are present in one core and not the other. Ideally the changes only impact the package file (but of course in reality more files will be impacted). In short the goal is to push differences to the package file as much as possible, enable maximum re-use, and ease code review by for example grouping all pipeline registers into structs like id_ex_pipe_t. 

Our own self imposed limitation is that we will not use interfaces on the top level interface itself (based on experience). Using structs and interfaces internal in the RTL should not cause issues to industry standard tools.

MikeOpenHWGroup, 2021-05-04:
> Reason to leave it there is just in case other people contribute, in which case some style guide is better than none.

Well, that's a rationalization if ever there was one, but who among us can get through a day without rationalizations?

My question is answered so I will close this issue.  If I get an response from the Ibex team, I'll update this issue just for the record.

closed 2021-05-04 by MikeOpenHWGroup