openhwfoundation/cv32e40x #52: Analyze branch bypass removal

silabs-halfdan, 2021-05-19:
Analyze effects of removing branch bypass and whether it affects design choices for the new controller.

The commit with bypass removed:
https://github.com/openhwgroup/cv32e40x/pull/51/commits/94c89ff77c13c291f7939082f5c6bd6138a45ac7

Silabs-ArjanB, 2021-05-19:
See https://github.com/openhwgroup/cv32e40x/pull/51

silabs-oysteink, 2022-01-21:
The change here was that 'branch_in_ex' was removed from 'ex_ready_o'.
For the E40P, this allowed branches to retire from the EX stage without having to propagate to the WB stage. This could potentially lead to a small performance gain as instructions after the branch were allowed to enter EX even though the WB stage was not  yet ready.

For the E40X and E40S, we have a new controller, and all instructions retire from the WB stage. This makes it impossible to retire branches from EX as above, and thus this change would have been needed when the new controller was introduced anyway.
[image]

closed 2022-01-27 by silabs-oysteink