openhwfoundation/cv32e40x #68: User Manual: PMA Address Ranges

eroom1966, 2021-06-07:
The address range specification in section 7.1? This doesn't look right. It says:

The address boundaries of a PMA region are set in word_addr_low/word_addr_high. These contain bits 33:2 of 34-bit, word aligned addresses. To get an address match, the transfer address addr must be in the range word_addr_low <= addr < word_addr_high.

As far as I can see this definition means that the address of the very top word of physical memory (0x3fffffffc) can never match a PMA region address. Is this right? Perhaps the comparison should be:

    {word_addr_low,2'b00} <= {addr[33:2],2'b00} < {word_addr_high,2'b11}

Silabs-ArjanB, 2021-06-07:
Hi @eroom1966 The address range definition matches the TOR definition of the official RISC-V PMP. I agree that the formula used there in and our user manual is not clear enough. I propose to change it to the following instead (which differs from what you wrote above):

{word_addr_low,2'b00} <= addr[31:0] < {word_addr_high,2'b00}

JamesKenneyImperas, 2021-06-07:
To elaborate a couple of things from eroom1966's post:

1. I think the address being compared should remain as `{addr[33:2],2'b00}` or be `addr[33:0]`, not `addr[31:0]`. In general, the input is a 34-bit address, not 32-bit (when S-mode is implemented). I'm not sure if S-mode is relevant to this implementation.
2. The problem with using the TOR matching definition from the base specification is that there is no way using this match algorithm to specify permissions for the very top word of 34-bit physical memory (address `0x3fffffffc`). For PMP regions this can be worked around using a different region type (NA4 or NAPOT) but those aren't available here.

Thanks.

Silabs-ArjanB, 2021-06-08:
Hi @JamesKenneyImperas The CV32E40X does not implement S-mode. Where in the RISC-V specification do you find that addresses are supposed to be 34 bit (the PC in this implementation is 32-bit)?

Silabs-ArjanB, 2021-06-08:
I think I found it and we can document it as a 34-bit address with the remark that 32-bit addresses are always zero-extended to 34-bit addresses.

JamesKenneyImperas, 2021-06-08:
Ok - but are you in future intending to use this same static configuration mechanism on processors that also implement S-mode? In that case, it will not be possible to enforce anything other than default access to the top word of physical memory.

Silabs-ArjanB, 2021-06-08:
Hi @JamesKenneyImperas If we would implement S-mode (which we will not do for the 40X and 40S), then it would be logical to extend the PMA configuration to also support NA4 or NAPOT, just like the PMP. For now I do not see a reason to support such configurations.

Silabs-ArjanB, 2021-06-08:
Hi @eroom1966, @JamesKenneyImperas 

Are you okay if we change the User Manual to the following:

> 
> {word_addr_low,2'b00} <= addr[33:0] < {word_addr_high,2'b00}
>  
> Note that addr[33:32] = 2'b0 as the CV32E40X does not support Sv32.

JamesKenneyImperas, 2021-06-08:
That looks accurate to me, given the previous discussion.

Silabs-ArjanB, 2021-06-14:
Fixed with https://github.com/openhwgroup/cv32e40x/pull/74

closed 2021-06-14 by Silabs-ArjanB