openhwfoundation/cv32e40x #69: Confirm exception for misaligned accesses to PMA I/O regions

JamesKenneyImperas, 2021-06-08:
The CORE-V-Docs Documentation states in section 7.2 (Main memory vs I/O):

> An attempt to perform a non-naturally aligned load access to an I/O region causes a precise load access fault (exception code 5). An attempt to perform a non-naturally aligned store access to an I/O region causes a precise store access fault (exception code 7).

Can I just quickly confirm that these should not be load address misaligned (4) and store/AMO address misaligned (6)? Thanks.

Silabs-ArjanB, 2021-06-08:
Yes, we use code 5 and 7 on purpose here as opposed to the misaligned faults (4 and 6), which are meant for a different purpose. 

Don't you agree that this is what is intended in the official RISC-V specs?

JamesKenneyImperas, 2021-06-08:
I think this is due to a major internet outage

Silabs-ArjanB, 2021-06-08:
An EEI may not guarantee misaligned loads and stores are handled invisibly. In this case, loads
and stores that are not naturally aligned may either complete execution successfully or raise an
exception. The exception raised can be either an address-misaligned exception or an access-fault
exception. For a memory access that would otherwise be able to complete except for the misalignment,
an access exception can be raised instead of an address-misaligned exception if the misaligned
access should not be emulated, e.g., if accesses to the memory region have side effects. When an
EEI does not guarantee misaligned loads and stores are handled invisibly, the EEI must define if
exceptions caused by address misalignment result in a contained trap (allowing software running
inside the execution environment to handle the trap) or a fatal trap (terminating execution).

JamesKenneyImperas, 2021-06-08:
Ok, I'll close this. The bottom line is that this is an implementation choice, I think.

closed 2021-06-08 by JamesKenneyImperas