openhwfoundation/cv32e40x #71: Documentation: NMI Interrupts for Bus Load/Store faults

eroom1966, 2021-06-10:
In the documentation for bus load/store faults, these are described as Non Maskable, however in the interrupts section 14.2 it says

> After reset, all interrupts are disabled.

> To enable interrupts, both the global interrupt enable (MIE) bit in the mstatus CSR and the corresponding individual interrupt enable bit in the mie CSR need to be set

I am guessing this is ambiguous and that 
A. load/store bus fault is NOT disabled after interrupt
B. load/store bus fault is NOT enabled by the global interrupt enable (MIE)

Also, the document says the following 

> All interrupt lines are level-sensitive

Is this correct ?
If NMI is truly level-sensitive AND it cannot be disabled by the interrupt enable MIE, then immediately upon taking the NMI it will trigger another NMI if it is still (level) active, creating a lockup, unable to execute any code.

Thx
Lee

Silabs-ArjanB, 2021-06-10:
Hi @eroom1966 Thank you for reporting this. I will update section 14.2 to reflect that NMIs are always enabled (also right after resest) and that MIE does not impact them. With respect to the level-sensitivity, I think that remark is correct. The remark is about the interrupt lines (I will add the word 'irq_i' there), so it does not relate to NMIs.

Silabs-ArjanB, 2021-06-10:
Hi @eroom1966 Could you please review https://github.com/openhwgroup/cv32e40x/pull/72 and comment if that addresses your concern?

eroom1966, 2021-06-10:
Hi @Silabs-ArjanB 
Reading the changes in #72, level-sensitive will apply to the irq_*
I think we need to explicitly say that the NMI are edge sensitive, as they cannot be disabled.
I will comment on the review
Thx
Lee

Silabs-ArjanB, 2021-06-10:
Hi @Silabs-ArjanB, I think using the word edge or level sensitive is a bit weird in the context of our NMIs as there are no corresponding NMI pins on the core. I could describe the NMIs being triggered by certain 'load/store fault events', which I think then would convey the same meaning as what you might call edge sensitive.

Silabs-ArjanB, 2021-06-11:
Fixed in https://github.com/openhwgroup/cv32e40x/pull/72

closed 2021-06-11 by Silabs-ArjanB