openhwfoundation/cv32e40x #73: Question: confirm that dcsr[nmip] is always 0

eroom1966, 2021-06-10:
Can I confirm from the documentation that dcsr[nmip] is always 0
If this is the case then when in debug mode (NMI Disabled), a load/store bus fault error condition cannot be detected ?

Silabs-ArjanB, 2021-06-10:
No, nmip can be 1. All content in the User Manual related to Debug is a copy from the CV32E40P for now and all of it will need to be updated to the additional features that will be present on the CV32E40X (and CV32E40S), e.g. parameterized number of triggers, implementation of _load_ bitfield, _store_ bitfield, etc. Also we will aim to be compliant with version 1.0 (we will be working to be compatible with Version 1.0.0-STABLE, 090eac8b85ba90c71d15773e783eaa3b527e5d8d and we are assuming it will get ratified into 1.0 in time). We can prioritize updating the User Manual accordingly.

Silabs-ArjanB, 2021-06-10:
The debug features that we intend to add are also described in section 3.8 of the 'CV32E40X, CV32E40S - ISS SUPPORT' document that I sent to you earlier.

Silabs-ArjanB, 2021-06-11:
Fixed in https://github.com/openhwgroup/cv32e40x/pull/72

closed 2021-06-11 by Silabs-ArjanB