openhwfoundation/cv32e40x #124: Investigate if halting EX is SEC clean

silabs-oysteink, 2021-07-02:
# Halting of execute stage

Commit 436d2b1b768f1460a98aa4f36aafd1f7db8d8137 incorporates the ctrl_fsm.halt_ex into the execute stage's local instr_valid signal. This was done to be able to stall the execute stage to read the correct minstret value.

The SEC status of this change is currently not know, as the tool is running for a very long time. This must be revisited and updates once the run completes.

silabs-oysteink, 2021-08-17:
I ran a SEC run with the following hashes:
Commit before adding halt_ex to ex stage local instr_valid: 8c239922d2e82ec88d4023e2966b1cb72b6ac779
Commit after adding halt_ex to ex stage local instr_valid:    436d2b1b768f1460a98aa4f36aafd1f7db8d8137

With no changes to the checked out RTL, the SEC run runs for a very long time and eventually gets a timeout after 7 days (tool setting).
When changing the '*' operator in the multiplier to a '+' instead, the two RTL versions prove to be SEC equivalent after about 1,5 hours.

Silabs-ArjanB, 2021-08-17:
Perfect

closed 2021-08-17 by Silabs-ArjanB