openhwfoundation/cv32e40x #129: Move decoding from ALU to decoder

Silabs-ArjanB, 2021-07-02:
Commit fc8f68d introduced decoding logic into the ALU that really belongs in the decoder, for example: https://github.com/openhwgroup/cv32e40x/blob/fc8f68d80bd89e72ba91d930f7cf4210ce2a2c84/rtl/cv32e40x_alu.sv#L142

Now we are first _encoding_ in the decoder and then again _decoding_ in the ALU. Please check the effect of this transformation area wise relative to this commit.

silabs-oivind, 2021-07-09:
Agree that this logic belongs in the decoder. 
But, before we move it, we should clean up how this shifter is controlled. 
The current control signals (shifter_arithmetic, shifter_rotate, shifter_single_bit, shifter_rshift) are overloaded, and/or the names are misleading.
For instance, bset (bit set) is performed by setting shifter_single_bit=1, shifter_rotate=1. While bclr (bit clear) is performed by setting shifter_single_bit=1, shifter_arithmetic=1.

Silabs-ArjanB, 2021-07-28:
I did various experiments on this and ended up with https://github.com/openhwgroup/cv32e40x/pull/151

Removed shifter_arithmetic, shifter_rotate, shifter_single_bit, shifter_rshift as they were indeed misleading. Picked alu_operator in decoder such that decode are was minimized and such that critical info comes from a single bit.

closed 2021-07-28 by Silabs-ArjanB