openhwfoundation/cv32e40x #160: eXtension Interface Review Feedback

vogelpi, 2021-08-02:
Hi @Silabs-ArjanB ,

I did review [your updated eXtension interface](https://cv32e40x-user-manual.readthedocs.io/en/latest/x_ext.html). In general, this looks good and I think the main limitation of the [original CVXIF spec](https://github.com/openhwgroup/core-v-xif) on back to back memory ops etc. are mitigated with your proposal.

Below, I have a list of remarks/questions regarding your proposal. I've posted them here as your proposal currently lives in this repo. Ultimately, it would be great if your proposal got merged into the CVXIF spec.

Compressed Instructions:
1. In case of compressed instructions, why does the core need the uncompressed instruction in the first place? Isn't it possible to to identify the rs/rd indices already from the compressed instruction?
2. Your proposal says "The instruction is originally compressed and the coprocessor accepted the compressed instruction and provided a 32-bit uncompressed instruction. In this case the 32-bit uncompressed instruction will be attempted for offload even if it matches in the core’s non-compressed instruction decoder." Does that mean it would be legal for a coprocessor to have instructions with uncompressed encodings overlapping with other instructions, e.g., of the base instruction set? Isn't that dangerous? Is this really necessary or did I misunderstand something?
3. Signal dependencies: In your proposal you note that CV32E40X splits the compressed/uncompressed decoders with a pipeline stage to avoid outputs depending directly on inputs. For simpler and configurable cores like Ibex and if one doesn't target max frequency, this might not be worth the complexity. Meaning such cores wouldn't comply with the spec. Should we make an exception here for simple cores? Or weaken the requirement into a recommendation?

Memory Transactions:
1. Should we specify load/store support on the core side as optional feature for the extension interface? Obviously for CV32E40X you will have it but I can imagine that the most simple cores/coprocessors don't need this.
2. For unaligned loads/stores (split into two aligned transactions) a PMP look-up is required for both aligned transactions. This means a synchronous PMP fault can happen in at least two cycles. I think the proposal and also the original spec currently can't handle this. Your proposal assumes that the fault is signaled in `x_mem_resp_t` during the `x_mem_valid/ready` handshake. We should probably discuss this.

Handshakes:
1. Some typos "The signals in x_issue_resp_i are valid when x_issue_req_o and x_issue_resp_i are both 1. There are no stability requirements." I think this should read "The signals in x_issue_resp_i are valid when x_issue_req_o and x_issue_ready_i are both 1. There are no stability requirements."
2. Two interfaces don't have a proper handshake: `x_commit_valid_o`/`x_mem_result_valid_o`. I assume this is to not make the core overly complex and I agree with this choice. But maybe we should discuss this in the working group and think if this can cause difficulties in the coprocessor that we didn't think of yet.
3. In the `Handshake rules` section (a space is needed between "with" and "``") 
4. Your proposal specifies that valid can be dropped without ready in case an instruction gets killed (I agree with this). The next sentence then says: "A new transaction can be started by changing the id signal and keeping the valid signal asserted (thereby possibly terminating a previous transaction before it completed)." I would strictly specify that this is only possible due to the killl signal either. Or did you have some other scenario in mind here?

General remarks:
1. It's not 100% clear to me whether all interfaces are in-order or can be out-of-order. Your spec says "A coprocessor is allowed to provide results to the core in an out of order fashion." and at the same time "The transaction ordering on the result interface must correspond to the transaction ordering on the issue interface.".

Silabs-ArjanB, 2021-08-02:
Hi @vogelpi ,

Thank you very much for your extensive feedback.

> In case of compressed instructions, why does the core need the uncompressed instruction in the first place? Isn't it possible to to identify the rs/rd indices already from the compressed instruction?

It is possible, but the compressed instruction format does not have very uniform rules about the location of the rs/rd indices. So, unless we want to mandate a specific encoding (which I prefer not to do), I thought it would be easier to leave it completely up to the coprocessor and only mandate the location of the rs fields in the uncompressed instruction.

> Your proposal says "The instruction is originally compressed and the coprocessor accepted the compressed instruction and provided a 32-bit uncompressed instruction. In this case the 32-bit [...]

Yes, that is what I mean.

> Isn't that dangerous? Is this really necessary or did I misunderstand something?

I don't see a danger in it, but I also don't see a strong need for it either (neither do I see a strong need to offload compressed instructions to begin with given that the free opcode space there is quite limited). 

What would you propose should happen then in case the core offloads an unrecognized compressed instruction and the coprocessor returns a 32-bit instruction which in itself is valid for the CPU? For sure the CPU should not treat that uncompressed instruction itself; it should either offload it or raise some kind of error (most likely illegal instruction fault even though it recognizes the uncompressed instruction).

> Signal dependencies: In your proposal you note that CV32E40X splits the compressed/uncompressed decoders with a pipeline stage to avoid outputs depending directly on inputs. For simpler and [...]

As far as I can see also Ibex has the predecoder in IF and the regular decoder in ID. If this is the case, then I would like to keep the restriction in place (as it does not hinder any current core and it is a simple and clear requirement).

> Should we specify load/store support on the core side as optional feature for the extension interface? Obviously for CV32E40X you will have it but I can imagine that the most simple cores/coprocessors don't need this.

Agreed. I will make it an optional feature of a CPU (and then if a CPU does not implement it, it will limit the type of coprocessors that you can connect. I will also make the compressed interface optional then (as many coprocessors would not use it).

> For unaligned loads/stores (split into two aligned transactions) a PMP look-up is required for both aligned transactions. This means a synchronous PMP fault can happen in at least two cycles. I think the proposal and also the original spec currently can't handle this. 

A PMP fault can indeed happen in both cycles. I am not sure why you state that this cannot be handled. In the current proposal the CPU simply informs the coprocessor of the access faults and the CPU handles it further internally, although I did not really specify that I see. I did specify that no memory result handshake shall happen for memory accesses that led to synchronous exceptions; I did not specify whether a regular result handshake should happen in this case. Any opinion on that?

> Your proposal assumes that the fault is signaled in x_mem_resp_t during the x_mem_valid/ready handshake. We should probably discuss this.

Yes, that is what I specified. That does not limit the core to handle the exception internally though.

> Some typos "The signals in x_issue_resp_i are valid when x_issue_req_o and x_issue_resp_i are both 1. There are no stability requirements." I think this should read "The signals in x_issue_resp_i are valid when x_issue_req_o and x_issue_ready_i are both 1. There are no stability requirements."

Will fix.

> Two interfaces don't have a proper handshake: x_commit_valid_o/x_mem_result_valid_o. I assume this is to not make the core overly complex and I agree with this choice. But maybe we should discuss [...]

Of course we can discuss this. The reason is indeed to limit complexity and area in the CPU (and possibly core interconnect).

> In the Handshake rules section (a space is needed between "with" and "``")

Will fix.

> Your proposal specifies that valid can be dropped without ready in case an instruction gets killed (I agree with this). The next sentence then says: "A new transaction can be started by changing the [...]

If the CPU changes it mind during the offload via the issue interface, then the id signals can change while valid remains asserted; in such scenario although the offload attempt is killed by the CPU, the actual kill signal is not used for this.

Silabs-ArjanB, 2021-08-02:
Easy feedback (not requiring further discussion) has been addressed via https://github.com/openhwgroup/cv32e40x/pull/161

vogelpi, 2021-08-02:
Thanks for the quick reply and further explanations @Silabs-ArjanB . For simplicity, I am only replying to questions that you didn't resolve already.

> It is possible, but the compressed instruction format does not have very uniform rules about the location of the rs/rd indices. So, unless we want to mandate a specific encoding (which I prefer not [...]

I wasn't really familiar with compressed instructions and e.g. wasn't aware that there weren't strict rules on the rs/rd fields. Thanks for providing more details on your motivation here, that's very useful. I totally agree that instructions uncompressed by a coprocessor should be offloaded not matter whether they match with an internal uncompressed instruction or not. I see this as the only option. 

> As far as I can see also Ibex has the predecoder in IF and the regular decoder in ID. If this is the case, then I would like to keep the restriction in place (as it does not hinder any current core [...]

You are right. I completely forgot about the compressed decoder being in IF also in Ibex. No need for weakening the requirements. 

> A PMP fault can indeed happen in both cycles. I am not sure why you state that this cannot be handled. In the current proposal the CPU simply informs the coprocessor of the access faults and the CPU [...]

Yes, you are right, the exception can be handled by the core internally. But what should the coprocessor do in that case? Let's assume we do an unaligned load where the first load transaction passes PMP. Hence the x_mem_resp_t doesn't signal a synchronous exception. But then the second one does trigger a PMP fault. This can no longer be signaled through x_mem_resp_t because the handshake already happened right? So the core would need to abort the instruction in the coprocessor either with the kill signal, or through x_mem_result_o (right now not specified to happen for transactions causing a sync exception). I think being able to signal a synchronous exceptions via two different paths should be avoided for simplicity.

Or would the idea for the core to wait with the x_mem_resp_o/x_mem handshake until both aligned transactions have been checked/both transactions have been issued?

> If the CPU changes it mind during the offload via the issue interface, then the id signals can change while valid remains asserted; in such scenario although the offload attempt is killed by the [...]

Oh, I missed that this only holds for the issue interface. You are right, there is no kill signal for this. The kill signal is only meaningful for offloaded instructions. Your proposal makes sense.

Silabs-ArjanB, 2021-08-02:
Hi @vogelpi 

> Yes, you are right, the exception can be handled by the core internally. But what should the coprocessor do in that case? Let's assume we do an unaligned load where the first load transaction passes [...]

Correct. I missed that one. I was already doubting whether we should make the CPU responsible for splitting non-aligned transactions into multiple aligned transactions or not. Originally I thought that would be cleanest, but now I wonder if we should require that coprocessors take on this responsibility (so that a coprocessor needs to make sure that it only performs aligned memory requests to the CPU. For a misaligned offloaded load/store the coprocessor would then perform up to two memory requests to the CPU (so that it can get a memory response for each request separately).

> So the core would need to abort the instruction in the coprocessor either with the kill signal, or through x_mem_result_o (right now not specified to happen for transactions causing a sync [...]

In case of a synchronous exception on the first transfer attempt of a misaligned load/store, the CV32E40X will suppress the second transfer attempt of the same misaligned load/store no matter if that itself has an exception or not. This behavior I want to put in the specification as a CPU requirement. I am not sure whether the CPU should cause a kill of the related instruction in case this misaligned load/store is due to an offloaded instruction; I prefer not to cause an explicit kill here.

Currently we have this in the spec (_which maybe I should change_):

In case of a synchronous exception (associated with the  memory request transaction) no corresponding transaction will be performed over the memory result (x_mem_result_valid_o) interface.

It would be easier to mandate the following:

_For a memory request handshake always perform the corresponding memory result handshake and  result handshake (no matter if a synchronous exception happened or not). After a synchronous exception the coprocessor should simply ignore the related memory result handshake and on the corresponding result handshake it should use the exc/excode bits to inform the CPU that a synchronous exception happened (although the CPU could already have determined that by itself)._

I then also want to add the following as a CPU requirement:

_In case the CPU signals a synchronous exception (associated with the  memory request transaction) on any memory request transaction, then the CPU shall suppress all subsequent external bus interface memory transactions for the same offloaded instruction._

> Or would the idea for the core to wait with the x_mem_resp_o/x_mem handshake until both aligned transactions have been checked/both transactions have been issued?

I prefer to make the coprocessor responsible for only issuing aligned transactions.

Silabs-ArjanB, 2021-08-03:
Hi @vogelpi 

> If the CPU changes it mind during the offload via the issue interface, then the id signals can change while valid remains asserted; in such scenario although the offload attempt is killed by the [...]

The id value can be changed during _any_ not yet acknowledged handshake (not only the issue handshake) to retract an ongoing request and start with a new request. This is done to allow for easier interconnect implementation (e.g. for allowing broadcast of handshake signals).