openhwfoundation/cv32e40x #181: cleanup compiler warnings related to rvfi output

strichmo, 2021-08-14:
### Task Outcome
The current cv32e40x_wrapper module leaves the RVFI output signals from cv32e40x_rvfi are unconnected.  When compiling in simulation a huge amount of unconnected output warnings are generated.

```
xmelab: *W,CUVWSP (/work/strichmo/ip_rad_riscv/mdx_e40x/modules/core-v-verif/core-v-cores/cv32e40x/bhv/cv32e40x_wrapper.sv,260|11): 162 output ports were not connected:
xmelab: (/work/strichmo/ip_rad_riscv/mdx_e40x/modules/core-v-verif/core-v-cores/cv32e40x/bhv/cv32e40x_rvfi.sv,152): rvfi_valid
xmelab: (/work/strichmo/ip_rad_riscv/mdx_e40x/modules/core-v-verif/core-v-cores/cv32e40x/bhv/cv32e40x_rvfi.sv,153): rvfi_order
xmelab: (/work/strichmo/ip_rad_riscv/mdx_e40x/modules/core-v-verif/core-v-cores/cv32e40x/bhv/cv32e40x_rvfi.sv,154): rvfi_insn
xmelab: (/work/strichmo/ip_rad_riscv/mdx_e40x/modules/core-v-verif/core-v-cores/cv32e40x/bhv/cv32e40x_rvfi.sv,155): rvfi_trap
```

Currently the RVFI Agent from core-v-verif binds directly to the cv32e40x_rvfi module and no output ports are required for simulation.  

Since these outputs may be necessary for other contexts, I would like to see these cleaned up.

### Background information
Compile the core-v-verif testbench to see warnings.
% makeuvmt clean comp CV_CORE=cv32e40x

### Completion Criteria
Rerun the compilation above from clean and verify the warnings are not emitted.

silabs-halfdan, 2021-08-16:
Hi @silabs-robin,
You removed these hookups for a reason at some point, was that only needed for formal?
If so I'll just add an `else to the formal ifdef and assign these locally.

silabs-robin, 2021-08-16:
Are we talking about this? https://github.com/openhwgroup/cv32e40x/blob/master/bhv/cv32e40x_wrapper.sv#L108

AFAIK, this is how it goes;
riscv_formal uses 40x wrapper as dut, because the rvfi signals are needed.
RVFI_OUTPUTS and RVFI_CONN are macros that make this easy to connect.
These should only be present if RISCV_FORMAL is defined.

To my knowledge, formal is the only one using those signals/macros.
I don't know what the removed hookups are.
Is all we need something like this?:
```
`ifdef RISCV_FORMAL
         ,`RVFI_CONN
`else
       ,.rvfi_valid(),
       ,.rvfi_...
`endif
```

silabs-halfdan, 2021-08-16:
Ok good, that should be all that is needed. I will make a PR.

closed 2021-08-18 by Silabs-ArjanB