openhwfoundation/cv32e40x #182: instruction bus fault raises incorrect exception index

strichmo, 2021-08-14:
### Component

**Component:RTL**

The RTL appears to issue a mcause of 0x18 (24) instead of the quoted 48 from the CV32E40X User Manual.
https://cv32e40x-user-manual.readthedocs.io/en/latest/exceptions_interrupts.html?highlight=bus%20fault#exceptions

### Steps to Reproduce
1. Use cv32e40x/dev branch of core-v-verif
2. 28591cf6860c20d03355b1df3944b95f73a619f7
3. makeuvmt test TEST=instr_bus_error SEED=1 CV_CORE=cv32e40x USE_ISS=0

*Note the ISS is not quite ready so please run without it for now*

The test should get in an infinite loop and will need to be interrupted with ctrl-c.

```
In the word-aligned bus error func
CV32E40X BSP: unknown exception handler entered

CV32E40X BSP: unknown exception handler entered

CV32E40X BSP: unknown exception handler entered

CV32E40X BSP: unknown exception handler entered

CV32E40X BSP: unknown exception handler entered
```

Silabs-ArjanB, 2021-08-16:
At first glance this seems to be simply wrong in the package file. As part of the fix, please add all other 40X (not 40S) exception codes in that package file.

Silabs-ArjanB, 2021-09-08:
Should be fixed with https://github.com/openhwgroup/cv32e40x/pull/207

strichmo, 2021-09-14:
This is now fixed and tested in a directed test: instr_bus_error in the CV32E40X's core-v-verif.

closed 2021-09-14 by strichmo