openhwfoundation/cv32e40x #183: RVFI trace not going to its own file, inserted into Indago SmartLog

strichmo, 2021-08-15:
### Component
**Component:Other**: For any other issues

The tracer in the RVFI (cv32e40x_rvfi_trace.svh) is emitting $fwrite calls with a filehandle of x.  There should either be a proper file opened for this or a switch added to enable/disable rvfi tracing.

### Steps to Reproduce
Run any test.  In the debugger you should be able to see that the $fwrite calls are using an X filehandle.

You can also run Indago post-processing mode.  In SmartLog, the stdout message contains these extra RVFI trace messages.  Ideally these should be in their own file for easier filtering.

Silabs-ArjanB, 2021-08-16:
Also document the (new) behavior in the User Manual.

silabs-halfdan, 2021-08-18:
This ticket should now be resolved.

strichmo, 2021-09-07:
Confirmed that the RVFI log is emitted into its own file.

closed 2021-09-07 by strichmo