openhwfoundation/cv32e40x #187: RVFI may set dpc before entering debug handler during single-step mode

strichmo, 2021-08-17:
### Component
**Component:RVFI**
**Component:Doc **: For issues in the Documentation (e.g. for User Manual or README.md files)
**Component:Other**: For any other issues

### Steps to Reproduce
Please provide:
1. e422fa37d0f5c50e526dbfddb01f927c7d1fd513 (cv32e40x/dev branch)
2. Using Xcelium 21.07.a001
```
%  makeuvmt test TEST=debug_test WAVES=1 ADV_DEBUG=1 SEED=1680260209 CV_CORE=cv32e40x
```

The RVFI will report the dpc CSR updating to the "next" instruction during the instruciton retirement of the stepped instruction (0xca94).  However the ISS will always wait until the first instruction retirement in the debug handler before updating dpc to the next instruction (0xca98).  This behavior is intermittent-not all single-stepped instructions show this behavior on RVFI.

[image]

silabs-halfdan, 2021-08-19:
@silabs-oysteink will implement a fix for this in RTL.

Silabs-ArjanB, 2021-08-19:
Hi @silabs-oysteink , this also touches on the discussion we just had. Here you could also argue that all instructions should set dpc as part of their execution in single step mode.

silabs-oysteink, 2021-08-19:
Agreed @Silabs-ArjanB, there will be a PR shortly which fixes this and also aligns on CSR update when the stepped instruction retires.

Silabs-ArjanB, 2021-08-20:
Should be fixed with https://github.com/openhwgroup/cv32e40x/pull/192

strichmo, 2021-09-07:
Confirmed fixed.  The debug_test is passing robustly with and without random stalls.

closed 2021-09-07 by strichmo