openhwfoundation/cv32e40x #195: DSIM compiler failure with PMA_NUM_REGIONS=0

strichmo, 2021-08-20:
### Steps to Reproduce
Compile the head of master of the cv32e40x RTL on Metrics DSIM.

With the default PMA_NUM_REGIONS setting of 0, the PMA_CFG array is configured with a -1:0 range.  
https://github.com/openhwgroup/cv32e40x/blob/adcb717bce94386c47feb7b7179ec75a0a381d87/rtl/cv32e40x_core.sv#L37

This needs to be fixed to ensure a 0:0 range via a conditional in the range derivation.

For reference a similar fix was applied to the CV32E40S to fix the same issue with DSIM.
https://github.com/openhwgroup/cv32e40s/pull/38

This is blocking a PR on the core-v-verif so please let @strichmo know when this is ready to be tested.  https://github.com/openhwgroup/core-v-verif/pull/822

Silabs-ArjanB, 2021-08-23:
@strichmo Note that https://github.com/openhwgroup/cv32e40s/pull/38 also has negative ranges and was deemed okay. After that PR the 40X was changed in a similar way (it did not have negative ranges and negative ranges (using int instead of unsigned) were re-introduced to get a uniform and cleaner coding style). This was done in https://github.com/openhwgroup/cv32e40x/pull/191, but obviously something else goes wrong now due to the nagative ranges.

Can you share the log file so that we can see what goes wrong?

silabs-oivind, 2021-08-23:
@strichmo Could you try updating the PMA parameters in the testbench in the same way as the RTL was updated in #191 ?

strichmo, 2021-08-23:
I have updated the testbench to ensure a consistent signed int in all of the parameter derivations.  It appears that DSIM is OK with this so I will update the pull request.

strichmo, 2021-08-23:
This issue can be closed.

closed 2021-08-23 by Silabs-ArjanB