openhwfoundation/cv32e40x #239: document behavior of core with respect to OBI data bus errors and NMI handlers

strichmo, 2021-09-21:
### Component

**Component:Doc **: This bug is intended to discuss and document implementation of NMI handlers and OBI data bus errors to enable proper modeling and verification of the functionality.

### Issues

The following sections are behaviors that need to be properly defined to ensure accurate modeling for verification.  In this initial comment, each section will be updated with the understood functionality.  Please comment on this Issue to provide updates, clarifications, etc.  In the final comment of this Issue there should be a restating of each section with the agreed-upon descriptions for each section.

#### Imprecise timing of NMI entry

The actual timing of NMI handler entry with respect to the detection of the OBI bus error is not fully deterministic.  

* In debug mode, the NMI will not be entered until a dret instruction is executed, or single-stepping when dcsr.stepie is set.
* In non-debug mode, the NMI will take some number of instructions before entry.  **Design should provide an upper-limit on the number of instruction retirements from an OBI bus error to NMI handler entry**

#### Multiple NMI

In the case of multiple OBI bus errors, the RTL will "sticky" the mcause register (as a pending NMI) based on the first detected OBI error.  Upon NMI handler entry, the RTL can then latch as pending a subsequent OBI bus errors.  Any OBI bus errors detected while an NMI is pending will be dropped.

#### dcsr.nmip

This CSR register bit will set whenever a pending NMI is detected based on an OBI bus error, *regardless of being in machine mode or debug mode.*

#### mepc, mstatus on NMI entry

(We are including this section, as the privileged specification is not clear with respect to mstatus handling upon NMI entyr.)

Upon entry into an NML, the mepc and the mstatus registers will be updated as if a machine-mode exception handler were entered:

```
mepc <- PC[next]
mstatus.mpie <- mstatus.mie
mstatus.mie <= 0
```

Upon execution of mret from an NMI handler, the following will occur:
```
pc <- mepc
mstatus.mie <- mstatus.mpie
mstatus.mpie <- 1

Silabs-ArjanB, 2021-09-23:
> Imprecise timing of NMI entry
>
> The actual timing of NMI handler entry with respect to the detection of the OBI bus error is not fully deterministic.
>
>     In debug mode, the NMI will not be entered [...]

Proposed change:

Just like normal interrupts the handling of NMIs is deferred to a point in time in which the core is interruptible. The non-interruptible (potentially long) periods are:

- While in debug mode.
- While single stepping with dcsr.stepie = 0 (note that while performing a single step the core is not in debug mode).

Furthermore the core is non-interruptible for certain short periods as  interrupts (including NMIs) are only taken on instruction boundaries (i.e. either an instruction is fully performed or none of its effects, including data interface accesses, are visible).

When not in debug mode and not single stepping with dcsr.stepie = 0, if an interrupt, including NMIs, is pending and enabled, then the core will stop issuing new instructions such that an interruptible time slot is guaranteed to occur.

The instruction that initially causes the NMI (i.e. the load or store that receives a data bus error) will always retire before an associated NMI is taken. In addition to that, _once the NMI is visible to the core_, based on the specific pipeline architecture,  at 0, 1, or 2 additional instructions can retire before the NMI is taken. In the current implementation the possible additional instructions can only be a loads or stores (or likely any offloaded instruction (this is TBD)). 

_Note that for bufferable stores a bus error and the associated NMI is allowed to become visible an arbitrary time after execution of such store and therefore an arbitrary number of instructions can retire between the bufferable store and the NMI handler entry._

Similar to irq_ack we can add an nmi_ack signal inside the core to indicate when an NMI is actually taken. If desired we can add an internal nmi_id[31:0] as well which will indicate the PC of the (initial) instruction causing an NMI. (We don't know for sure yet whether a non-0 mtval will be implemented).

_Please let us know whether you want the nmi_ack and/or nmi_id._

> Multiple NMI
> 
> In the case of multiple OBI bus errors, the RTL save as a pending NMI based the first detected OBI error. The mcause of this pending error will be the first detected OBI bus error (i.e. load or store). This latched mcause will be updated to the CSR upon NMI handler entry. After entering the NMI handler, the RTL can then latch as pending a subsequent OBI bus errors Any OBI bus errors detected while an NMI is pending will be dropped.

Correct. We could (but we won't) redefine the mcause value to indicate the number and type of bus errors that occurred before the NMI is taken. Also there is a non-ratified proposal for recoverable NMIs (used in Ibex) that we are not implementing (see https://github.com/riscv/riscv-isa-manual/issues/261).

> dcsr.nmip
> 
> This CSR register bit will set whenever a pending NMI is detected based on an OBI bus error, regardless of being in machine mode or debug mode.

Correct, but this is only visible to the verification environment because we have an RVFI interface probing into the core. From a user/software point of view it is a don't care whether we set dcsr.nmip while in machine mode (as that CSR is not accessible in machine mode).

> mepc, mstatus on NMI entry
> 
> (We are including this section, as the privileged specification is not clear with respect to mstatus handling upon NMI entyr.)
> 
> Upon entry into an NML, the mepc and the mstatus registers will be updated as if a machine-mode exception handler were entered:

Upon entry into an NML, the mepc and the mstatus registers will be updated as if a machine-mode _interrupt_ handler were entered:

> mepc <- pc_next
> mstatus.mpie <- mstatus.mie
> mstatus.mie <- 0

Yes, but with the remarks that pc_next does not necessarily point to the instruction immediately after the one that caused the NMI.

> Upon execution of mret from an NMI handler, the following will occur:
> 
> pc <- mepc
> mstatus.mie <- mstatus.mpie
> mstatus.mpie <- 1

Yes, but of course normally you would not perform an mret as NMIs are not considered recoverable. Once we move to 1.12 of the spec also mstatus.mprv will be set to 0.

strichmo, 2021-09-23:
Thanks for the write-up Arjan.

> Proposed change:
>
> Just like normal interrupts the handling of NMIs is deferred to a point in time in which the core is interruptible. The non-interruptible (potentially long) periods are:
>
> * While [...]

I don't believe that the nmi_ack will be necessary.  Any assertions and the ISS interface will use the RVFI (rvfi_intr + rvfi_csr[mcause]) to determine NMI entry point at instruction retirement of the first handler. 

Is it a fair assumption that the "first" OBI D error (read or write) will always determine the mcause vector for the NMI?  Are there any race conditions (or will there be with the introduction of the write buffer)?

> > Multiple NMI
> > In the case of multiple OBI bus errors, the RTL save as a pending NMI based the first detected OBI error. The mcause of this pending error will be the first detected OBI bus error (i.e. load or store). This latched mcause will be updated to the CSR upon NMI handler entry. After entering the NMI handler, the RTL can then latch as pending a subsequent OBI bus errors Any OBI bus errors detected while an NMI is pending will be dropped.
> 
> Correct. We could (but we won't) redefine the mcause value to indicate the number and type of bus errors that occurred before the NMI is taken. Also there is a non-ratified proposal for recoverable NMIs (used in Ibex) that we are not implementing (see [riscv/riscv-isa-manual#261](https://github.com/riscv/riscv-isa-manual/issues/261)).
> 
> > dcsr.nmip
> > This CSR register bit will set whenever a pending NMI is detected based on an OBI bus error, regardless of being in machine mode or debug mode.
> 
> Correct, but this is only visible to the verification environment because we have an RVFI interface probing into the core. From a user/software point of view it is a don't care whether we set dcsr.nmip while in machine mode (as that CSR is not accessible in machine mode).
> 

Good point, but since we check *every* CSR on every instruction retirement it would be good to agree and proceed with a uniform implementation regardless of debug mode.

> Upon entry into an NML, the mepc and the mstatus registers will be updated as if a machine-mode _interrupt_ handler were entered:
>
>
> Yes, but with the remarks that pc_next does not necessarily point [...]

Silabs-ArjanB, 2021-09-24:
> Is it a fair assumption that the "first" OBI D error (read or write) will always determine the mcause vector for the NMI? 

Yes.

> Are there any race conditions (or will there be with the introduction of the write buffer)?

No, it should really only depend on the first OBI data error.

Silabs-ArjanB, 2021-11-10:
Fixed with https://github.com/openhwgroup/cv32e40x/pull/304

Silabs-ArjanB, 2021-11-18:
Hei @strichmo , can this be closed now?

strichmo, 2021-12-07:
Any subsequent functional issues will be captured in separate issues.  I am closing this issue now.

closed 2021-12-07 by strichmo