openhwfoundation/cv32e40x #253: Is the extension interface not part of CV32E40X yet?

michael-platzer, 2021-10-01:
Hi,

according to the [README file](https://github.com/openhwgroup/cv32e40x/blob/master/README.md) of this project as well as the [documentation](https://cv32e40x-user-manual.readthedocs.io/en/latest/x_ext.html) the CV32E40X core "offers a general purpose extension interface" which allows to attach a coprocessor or similar extensions to the core "without the need to change the RTL of CV32E40X itself".

However, when scanning through the source code in the [`rtl/`](https://github.com/openhwgroup/cv32e40x/tree/master/rtl) directory I cannot find any sign of this extension interface. In fact, using grep it seems that none of the signals which should be part of the interface, such as `x_issue_valid_o`, or even typedefs such as `x_issue_req_t` are present in any file in this repository (except the documentation). It is mentioned in two issues though, #160 and #236, but apparently these only discuss the documentation of the interface.

Did I miss something or has this extension interface not been implemented yet?

Silabs-ArjanB, 2021-10-04:
Hello @michael-platzer , thank you for your interest in the CV32E40X. The extension interface has indeed not been implemented yet, but it is the plan to add it according to the definition in https://docs.openhwgroup.org/projects/openhw-group-core-v-xif/. 

Would you be interested in contributing to that?

michael-platzer, 2021-10-04:
Hi @Silabs-ArjanB , thanks for the clarification!

Yes, I would be interested in contributing.

Silabs-ArjanB, 2021-10-04:
Hi @michael-platzer , great! 

I see you are working on a open source vector coprocessor as well. It would be great to see if that would fit the eXtension interface and/or if we would need to change the eXtension interface to enable interfacing to a vector coprocessor.

The guidelines with respect to contributing can be found at https://github.com/openhwgroup/cv32e40x/blob/master/CONTRIBUTING.md

You could do a dummy/small pull request to see if the signing of the commits (-s) works properly.

Some time ago I typed up the intended top level interface. You could start with reviewing these against the current specification and then add them to the top level interface and the core wrapper. I don't have Verilog for the type definitions yet, so you could write those yourself and add them in https://github.com/openhwgroup/cv32e40x/blob/master/rtl/include/cv32e40x_pkg.sv

Here are the top level pins: 

 ```
   // CV-X-IF: Compressed interface

    output logic                                 x_compressed_valid_o,

    input  logic                                 x_compressed_ready_i,

    output x_compressed_req_t                    x_compressed_req_o, 

    input  x_compressed_resp_t                   x_compressed_resp_i,

 

    // CV-X-IF: Issue interface

    output logic                                 x_issue_valid_o,   

    input  logic                                 x_issue_ready_i,   

    output x_issue_req_t                         x_issue_req_o,     

    input  x_issue_resp_t                        x_issue_resp_i,    

 

    // CV-X-IF: Commit interface

    output logic                                 x_commit_valid_o,  

    output x_commit_t                            x_commit_o,        

 

    // CV-X-IF: Memory request interface

    input  logic                                 x_mem_valid_i,     

    output logic                                 x_mem_ready_o,     

    input  x_mem_req_t                           x_mem_req_i,       

    output x_mem_resp_t                          x_mem_resp_o,      

 

   // CV-X-IF: Memory result interface

    output logic                                 x_mem_result_valid_o,

    output x_mem_result_t                        x_mem_result_o,    

 

    // CV-X-IF: Result interface

    input  logic                                 x_result_valid_i,  

    output logic                                 x_result_ready_o,  

    input  x_result_t                            x_result_i,        
```

Let's do a pull request (PR) for the pins first and then in a next PR feed the interface to the proper submodule (pipeline stage). After that (or earlier if you want) we can talk about possible next steps.

Thanks!

Best regards,
Arjan

michael-platzer, 2021-10-04:
Hi Arjan,

yes indeed, my goal is to adapt our vector coprocessor to be compatible with your extension interface proposal and use it with CV32E40X as main core.

The top level interface pins you posted seem to match the documentation.

I see that most of the type definitions use [these parameters](https://docs.openhwgroup.org/projects/openhw-group-core-v-xif/x_ext.html#parameters), so I assume that this would best be implemented as a parameterized interface?

Best,
Michael

Silabs-ArjanB, 2021-10-04:
Hi Michael,

> I see that most of the type definitions use these parameters, so I assume that this would best be implemented as a parameterized interface?

Yes indeed. There are three parameters however that do not need to be exposed on the boundary of the CV32E40X:

- X_DATAWIDTH (to be renamed to X_DATA_WIDTH). This will always be 32. This parameter can therefore simply be made a localparam in the 'package file' in the include directory.
- X_NUM_FRS. There are no plans for the CV32E40X to have an internal F related registerfile, so this can also be made a localparam in the pacakge file (tied to 2)
- X_ID_WIDTH. I expect that a value of 2 or 3 will be large enough. We can make it a localparam as well.

The other parameters can be added as parameters to the top level RTL (and the wrapper file). The intention is to support all of them. (The fewer parameters we have the less work we need to do on the verification side).

Silabs-ArjanB, 2021-10-05:
Hi @michael-platzer ,

We will have some requirements on the eXtension interface implementation, the most important ones are:

- Offloaded single cycle compute (i.e. non load/store) instructions need to have the same performance (cycle count) as non-offloaded ALU instructions, so the result interface needs to feed directly into the registerfile bypass in the 40X.
- Offloaded load instructions must be able to have the same performance as non-offloaded load instructions (so the equivalent of the load-use stall is expected, but from the WB stage again the result interface needs to bypass the registerfile in the 40X).

michael-platzer, 2021-10-06:
Hi Arjan,

I wrote the initial code for the interface according to the spec and created a first pull request for it. Please let me know if this is what you had in mind.

Best,
Michael