openhwfoundation/cv32e40x #258: User Manual: Update the debug description

silabs-robin, 2021-10-04:
The user manual does not include new (planned) 40x debug capabilities. E.g. `DBG_NUM_TRIGGERS` etc.

Silabs-ArjanB, 2022-09-28:
Also fix the following statement:

> debug_req_i is the “debug interrupt”, issued by the debug module when the core should enter Debug Mode. The debug_req_i is synchronous to clk_i and requires a minimum assertion of one clock period to enter Debug Mode. The instruction being decoded during the same cycle that debug_req_i is first asserted shall not be executed before entering Debug Mode.

We don't want remarks as 'the instruction being decoded' as this is not externally visible. There just needs to be a remark somewhere how to enter debug immediately when coming out of reset / fetch enable = 1.

Also update the remark about 'one clock period' if no longer correct after fixing https://github.com/openhwgroup/cv32e40x/issues/341.

silabs-oysteink, 2023-09-26:
I belive the user manual is up to date with a good description of the debug features already.

Silabs-ArjanB, 2023-09-27:
@silabs-robin Can you please close this ticket if agreed? (The points I raised have been addressed.)

silabs-robin, 2023-09-27:
Closing based on:
Lots of commits with improvements to the debug documentation have been pushed since the issue got created.
Arjan reports his points have been addressed.
The particular example of DBG_NUM_TRIGGERS is fixed.

closed 2023-09-27 by silabs-robin