openhwfoundation/cv32e40x #259: lint warning: read from output modport

MikeOpenHWGroup, 2021-10-04:
### Component
This lint warning occurs on lines 98 and 145 of `cv32e40x_instr_obi_interface.sv`.

### Steps to Reproduce
1. Git hash of the RTL compiled for this issue is `5cd7acf`, but by inspection it is also still there on the current head of the master branch (`92c46fd`).
2. I only see this lint report using Metrics DSIM:
```
$ git clone git@github.com:openhwgroup/core-v-verif.git
$ cd core-v-verif/cv32e40x/sim/uvmt
$ make SIMULATOR=dsim sanity
```
### Other info:
Warning produced by DSIM:
```
=W:[ReadingOutputModport]:
    Illegal read from output modport item:    

    /home/mike/GitHubRepos/MikeOpenHWGroup/core-v-verif/cv32e40x_dev/core-v-cores/cv32e40x/rtl/cv32e40x_instr_obi_interface.sv:98:13     

    /home/mike/GitHubRepos/MikeOpenHWGroup/core-v-verif/cv32e40x_dev/core-v-cores/cv32e40x/rtl/cv32e40x_instr_obi_interface.sv:145:24    
```

silabs-oivind, 2021-10-05:
Hi @MikeOpenHWGroup.
Is this warning new? This file hasn't been changed since May.
Our tools do not have any complaints about this. Is reading modport outputs not allowed by the LRM?

Silabs-ArjanB, 2021-11-18:
@MikeOpenHWGroup Can this issue be closed? We do not see a synthesis issue here and as far as we know we are not violating any LRM rule (if we are, then of course we will fix this).

MikeOpenHWGroup, 2021-11-18:
It _probably_ can, but please give me a couple of days to discuss with Metrics.  The dsim team knows the LRM well and they must have had a reason for adding this elaboration warning.   I will update this issue when I heard back from them.

MikeOpenHWGroup, 2021-11-19:
I just heard back from Metrics on this topic:
> There is no explicit rule or statement in the SystemVerilog LRM that says reading an output modport is not allowed. DSim is applying the same rules for module ports to modports and is providing the warning to notify users that their access(s) to the modport do not conform to the direction they declared for it. If the author of the modport had intended for the signal in question to be both sampled and driven, then they would have (should have) specified the direction as inout or ref.

I will close the issue and discuss with the verification team as to whether we should suppress this warning at compile time.

closed 2021-11-19 by MikeOpenHWGroup