openhwfoundation/cv32e40x #305: rvfi_intr not rising upon entering NMI handler

silabs-robin, 2021-11-08:
### Description

The problem is that `rvfi_intr` doesn't go high when entering NMI handler.

### Steps to Reproduce

I first saw the problem in `core-v-verif/cv32e40x/sim/uvmt` when I ran `make test TEST=data_bus_error GUI=1 ADV_DEBUG=1 WAVES=1`.

It should also be possible to see this in formal.
Though I don't have perfect asserts/covers for this, I did write _some_ that are capable of producing traces that highlights the issue:
https://github.com/openhwgroup/core-v-verif/compare/cv32e40x/dev...silabs-robin:nmi_assert

Silabs-ArjanB, 2021-11-10:
Fixed with https://github.com/openhwgroup/cv32e40x/pull/309

silabs-robin, 2021-11-16:
I got around to testing this now. AFAICT it works as I'd expect and I haven't seen any problems with it.

closed 2021-11-16 by silabs-robin