openhwfoundation/cv32e40x #332: illegal CSR does not take exception

strichmo, 2021-12-01:
### Component
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

### Steps to Reproduce

1. Use this forked repo: https://github.com/strichmo/core-v-verif.git
2. 5459b37d5ac846fa9171ff4334cd219af4c810c6
3. Run the _illegal_ test with any seed

```
% makeuvmt test TEST=illegal
```

The test should fail on the subtest that takes an exception on an illegal CSR.
```
Generate a decoded CSR operation with invalid CSR
UVM_ERROR @ 134688.300 ns : uvme_cv32e40x_core_sb.sv(381) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 9300, pc: 0x000000f2, csr: mepc, rvfi = 0x0000011c, rvvi = 0x000000f2, mask = 0xffffffff
UVM_ERROR @ 134688.300 ns : uvme_cv32e40x_core_sb.sv(381) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 9300, pc: 0x000000f2, csr: mstatus, rvfi = 0x00001880, rvvi = 0x00001800, mask = 0xffffffff
UVM_ERROR @ 134697.300 ns : uvme_cv32e40x_core_sb.sv(305) uvm_test_top.env.core_sb [CORESB] PC Mismatch, rvfi_order: 9301, rvvi_order: 9301, rvfi.pc = 0x000000f6, rvvi.pc = 0x00000000
```

Silabs-ArjanB, 2021-12-01:
Hi @strichmo This should be fixed now. Could you give it a look?

strichmo, 2021-12-03:
This is fixed in the latest RTL hash.  This PR from core-v-verif integrates the RTL: https://github.com/openhwgroup/core-v-verif/pull/1044

I will close this issue.

closed 2021-12-03 by strichmo