openhwfoundation/cv32e40x #333: debug CSR access in M-mode does not take exception

strichmo, 2021-12-01:
### Steps to Reproduce

1. Use this forked repo: https://github.com/strichmo/core-v-verif.git
2. 5459b37d5ac846fa9171ff4334cd219af4c810c6
3. Run the _generic_exception_test_ test with any seed

```
% makeuvmt test TEST=generic_exception_test
```

The test should fail with an assertion that fires when accessing dcsr in M-mode.

```
        ##0 cov_assert_if.illegal_insn_i;
                                       |
xmsim: *E,ASRTST (/work/strichmo/ip_rad_riscv/mdx/modules/core-v-verif/cv32e40x/tb/uvmt/uvmt_cv32e40x_debug_assert.sv,334): (time 15148800 PS) Assertion uvmt_cv32e40x_tb.dut_wrap.cv32e40x_wrapper_i.u_debug_assert.a_debug_regs_mmode has failed (2 cycles, starting 15142800 PS)
UVM_ERROR @ 15148.800 ns : uvmt_cv32e40x_debug_assert.sv(336) reporter [CV32E40X_DEBUG_ASSERT] Accessing debug regs in M-mode did not result in illegal instruction
```

Silabs-ArjanB, 2021-12-01:
Hi @strichmo This should be fixed now. Could you give it a look?

strichmo, 2021-12-03:
This is fixed in the latest RTL hash.  This PR from core-v-verif integrates the RTL: https://github.com/openhwgroup/core-v-verif/pull/1044

I will close this issue.

closed 2021-12-03 by strichmo