openhwfoundation/cv32e40x #334: rvfi does not enable determination of obi bus error with interrupt

strichmo, 2021-12-01:
### Component
**Component:RVFI**

### Steps to Reproduce
1. Use this forked repo: https://github.com/strichmo/core-v-verif.git
2. 5459b37d5ac846fa9171ff4334cd219af4c810c6
3. Run the _corev_rand_instr_obi_err_ test with the following seed
```
% makeuvmt comp_corev-dv gen_corev-dv test` TEST=corev_rand_instr_obi_err SEED=1196715667
```

In this case the OBI bus instruction error occurs on the I-fetch of the first instruction of an interrupt vector.  Note that the current RVFI does not enable the RVVI driver to determine that an interrupt was taken.  If the testbench does not get the signal that an interrupt was taken, the ISS cannot be informed, thus the _mepc_ will be incorrect.

```
UVM_ERROR @ 2217.300 ns : uvme_cv32e40x_core_sb.sv(305) uvm_test_top.env.core_sb [CORESB] PC Mismatch, rvfi_order: 25, rvvi_order: 25, rvfi.pc = 0x0000ff64, rvvi.pc = 0x0000014e
UVM_ERROR @ 2217.300 ns : uvme_cv32e40x_core_sb.sv(381) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 25, pc: 0x0000ff64, csr: mepc, rvfi = 0x0000ff64, rvvi = 0x0000014e, mask = 0xffffffff
UVM_ERROR @ 2217.300 ns : uvme_cv32e40x_core_sb.sv(381) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 25, pc: 0x0000ff64, csr: mstatus, rvfi = 0x00001800, rvvi = 0x00001880, mask = 0xffffffff
```

Silabs-ArjanB, 2021-12-01:
@silabs-halfdan Please confirm or disconfirm that RVFI does not offer the required visibility. If we need extra signals, can you come up with a proposal?

Silabs-ArjanB, 2021-12-01:
Hi @strichmo,

>>In this case the OBI bus instruction error occurs on the I-fetch of the first instruction of an interrupt vector. Note that the current RVFI does not enable the RVVI driver to determine that an interrupt was taken.

So in this case the first instruction of the handler will have both rvfi_trap = != 0x0 (indicating the instruction bus error) and it will have rvfi_intr = 1 (showing either synchronous exception or interrupt, but based on the mcause rdata bits you can see that it is an interrupt). Note that interrupt side effects in RVFI are (supposed to be) signaled in between instructions (i.e. on rdata) whereas synchronous exceptions are (supposed to be) signaled with the excepted instruction (i.e. on wdata).

I did not check the waveforms, but if above info in this post is what is actually happening, what info are you missing? Also note that the irq_ack is still available in the RTL, but that might not even be needed for this scenario.

strichmo, 2021-12-03:
I changed core-v-verif to use this methodology.  So far it appears to resolve the issue.  

From uvma_rvfi_instr_mon.sv:
```
             bit [XLEN-1:0] csr_mcause = mon_trn.csrs["mcause"].rdata;

            if (csr_mcause[31]) begin
               mon_trn.insn_interrupt    = 1;
               mon_trn.insn_interrupt_id = { 1'b0, csr_mcause[XLEN-2:0] };
            end
```

The PR was merged here: 
https://github.com/openhwgroup/core-v-verif/pull/1044

closed 2021-12-03 by strichmo