openhwfoundation/cv32e40x #339: cannot determine cause of OBI data bus error in debug mode

strichmo, 2021-12-03:
### Component
**Component:RTL**

### Steps to Reproduce
1. Use the cv32e40x/dev branch of core-v-verif https://github.com/openhwgroup/core-v-verif.git
2. Use this hash: 7d0b8501dafdd82ac02f997c70d7beed463be2b3
3. Command line
```
% makeuvmt comp_corev-dv gen_corev-dv test TEST=corev_rand_data_obi_err_debug SEED=-625886553
```

The bug manifests as this failure in the test:
```
UVM_INFO @ 1069.800 ns : uvma_rvfi_instr_mon.sv(209) uvm_test_top.env.rvfi_agent.instr_monitor0 [RVFIMON] Debug NMIP
UVM_ERROR @ 4539.300 ns : uvme_cv32e40x_core_sb.sv(381) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 225, pc: 0x00100000, csr: mcause, rvfi = 0x80000081, rvvi = 0x80000080, mask = 0xffffffff
```

The failure is caused by an incorrect mcause from the deferred NMI from debug mode.  The bus error occurs here:

```
    1043.000 ns |   OBID |  W | 0002b040 |  f | 00000000 | 00 |    7 |   00 | ERR
```

Eventually an instruction shows the dcsr.nmip asserted via the RVIF CSR bus: 
```
UVM_INFO @ 1069.800 ns : uvma_rvfi_instr_mon.sv(209) uvm_test_top.env.rvfi_agent.instr_monitor0 [RVFIMON] Debug NMIP

    1070.000 ns | RVFI |      271 |     33 | 1a110848 | 7b0029f3 | M | x0  | 00000000 | x0  | 00000000 | x19 | 400000cb |  -- | -------- | -------- | NMI dp_init_done corev_rand_data_obi_err_debug_0.S 17179 - csrrs x19,dcsr,x0
```

However at the instruction above there is no way to determine the correct "cause" of the NMIP via RVFI.  The ISS needs to have the proper NMI bit _at the time of the nmip assertion_ so that it can internaly latch the mcause (which will be propagated to the actual CSR after retirement of the next _dret_).

silabs-oysteink, 2021-12-07:
I am not sure what is going on here, but here are some thoughts after looking at the case:

There is an unbuffered store bus error happening while in debug mode. The dcsr.nmip is set, and the actual NMI handler is taken when the core exits debug mode. I can see two signals in the ISS io section named "LoadBusFaultNMI" and "StoreBusFaultNMI". Short time after the actual write error, the LoadBusFaultNMI is asserted (see figure below), and short time after the handler is actually entered the StoreBusFaultNMI is asserted. For the whole sequence, there is only one single bus error.
[image]

As seen in the next figure, the signal StoreBusFaultNMI is asserted after the NMI handler has been taken:
[image]

Note that this issue also made me discover a true RTL bug (issue #343) that is related, but not relevant for this specific case.

Silabs-ArjanB, 2021-12-07:
Hi @strichmo We could of course add a custom signal that has the same timing as the nmip bit; would that do the trick? We could 'pretend' that this new signal is the state of some new custom CSR. For bufferable writes this bit could still come much later than the RVFI packet that signals completion of a bufferable write; wouldn't that be an issue? Or another way of asking the question, what is the ISS supposed to do with this bit (it will get the accurate info once the NMI is taken, but obviously you want to signal something to the ISS much earlier)?

strichmo, 2021-12-07:
OVPSIM latches the internal cause and the dcsr.nmip state _when dcsr.nmip is asserted from RVFI_.  At that same instruction retirement we need to know whether the nmip is load or store.  At the NMIP assertion, OVPSIM will latch the cause internally.  We would have to ask for a modeling change to "alter" the mcause for the NMIP when the actual NMI is taken.

Silabs-ArjanB, 2021-12-08:
@strichmo We will add a rvfi_nmip[1:0] output reporting some internal CPU state:

rvfi_nmip[0] = nmi_pending_q;
rvfi_nmip[1] = nmi_is_store_q;

This will make the pending NMI visible in the RVFI packet for the instruction following the load or non-bufferable store that causes an NMI. For bufferable stores the reporting can be even later.

silabs-robin, 2022-01-03:
This issue should be ok to close now that https://github.com/openhwgroup/core-v-verif/pull/1087 is merged.

closed 2022-01-03 by Silabs-ArjanB