openhwfoundation/cv32e40x #340: dcsr cause mismatch when entering debug after a step

strichmo, 2021-12-03:
### Component
**Component:RTL**

### Steps to Reproduce
Please provide:
1. Use the cv32e40x/dev branch of core-v-verif https://github.com/openhwgroup/core-v-verif.git
2. Use this hash: 7d0b8501dafdd82ac02f997c70d7beed463be2b3
3. Command line
```
% makeuvmt comp_corev-dv gen_corev-dv test TEST=corev_rand_instr_obi_err_debug SEED=-1960056061 
```

silabs-oysteink, 2021-12-06:
Here is my understanding of the problem

1: There is an LSU instruction being stepped, this would normally lead to debug mode with cause=0x4 once the instruction finished with data_rvalid/wb_valid.
2: In the same cycle as wb_valid goes high (due to data_rvalid), we also get a debug_req_i
3.1: RTL decides to go to debug, with cause=0x3 for external request, as that is higher priority than single step (0x4)
3.2: rvfi_valid goes high for the stepped LSU instruction
5:  In debug mode, we get rvfi_valid for the first instruction of the handler, with rvfi_dbg = 0x3
6: haltreq is signalled to the ISS based on the last rvfi_valid with cause 0x3

I'm not yet sure how to solve this, but I believe that the RTL is doing the correct thing.
[image]
.

Silabs-ArjanB, 2021-12-06:
I think the root cause here is that the ISS should not assume that debug is entered due to single step even if a single stepped instruction completed. Only on the next instruction after that can the ISS know what the highest priority debug entry reason was.

strichmo, 2021-12-06:
@Silabs-ArjanB @silabs-oysteink Thank you for the explanations and additions.  From an ISA/debug spec perspective I agree with you and will forward this case to Imperas.

eroom1966, 2021-12-07:
To run this testcase it appears I also have to set the following on the make command
CV_CORE=cv32e40x 
Can you confirm if this is correct ?

silabs-robin, 2021-12-07:
> To run this testcase it appears I also have to set the following on the make command CV_CORE=cv32e40x Can you confirm if this is correct ?

Yes, you need that CV_CORE env variable set to "cv32e40x".

We have scripts that set up our environment, so I tried to find a minimal setup just now (I'm using tcsh though):
```
setenv CV_CORE cv32e40x

setenv CV_SW_TOOLCHAIN <point to gcc>
setenv CV_SIMULATOR <set to "xrun" or some other simulator>

setenv CV_SIM_PREFIX <if you need some job control prefix>
setenv CV_TOOL_PREFIX <if you need some job control prefix>

setenv IMPERASD_LICENSE_FILE <you might need to set this too>
setenv IMPERAS_QUEUE_LICENSE 1
setenv USE_ISS YES
```
I don't know 100% what is extraneous vs essential but this should work.

Then I `cd`'d into `cv32e40x/sim/uvmt` where I ran `make comp_corev-dv gen_corev-dv test TEST=corev_rand_instr_obi_err_debug SEED=-1960056061`.
(You could use "makeuvmt" like originally posted, but I prefer cd'ing and using "make".)

eroom1966, 2021-12-07:
how many instructions does it take for this test to complete ?

JamesKenneyImperas, 2021-12-08:
I think the behavior here is not defined sufficiently clearly in the Debug Specification. I have opened an issue there:

https://github.com/riscv/riscv-debug-spec/issues/691

Thanks.

eroom1966, 2021-12-08:
> I think the root cause here is that the ISS should not assume that debug is entered due to single step even if a single stepped instruction completed. Only on the next instruction after that can the ISS know what the highest priority debug entry reason was.
> 
> We could potentially add another rvfi_trap reason to signal that 'an instruction was single step (i.e retired) but a debug_request was honored as well determining the cause of the next debug entry'.

@Silabs-ArjanB do you think this is still the case given the description above by @JamesKenneyImperas ?

Silabs-ArjanB, 2021-12-08:
Hi @eroom1966 Yes I do. I don't understand the reasoning given in the above Debug Spec issue (but I will read it once more, because I did not spend much time on it and I can easily have misunderstood the intent. To me the concepts of 'after an instruction' and 'before an instruction' seem newly introduced to the Debug Spec, so at first glance I think the proposal is not backward compatible).

JamesKenneyImperas, 2021-12-08:
Hello @Silabs-ArjanB,

Triggers that activate before and after an instruction have been in the specification for a long time, I think - this is controlled by the `timing` field in the trigger.

To give a concrete example of where the specification is inadequate, consider the following simultaneous conditions. Note that "haltreq" in this context means that signal is asserted _during_ execution of an instruction when `dcsr.step=1` (if it is asserted before, then of course there is no problem - a debug exception with cause 3 will be taken before the stepped instruction executes).

1. dcsr.step on completed instruction + trigger before next instruction
In this case, the specification section 4.4.1 implies that step exception must be taken (cause 4)

2. haltreq + trigger before next instruction
In this case, priorities indicate that the higher-priority trigger must be taken (cause 2)

3. dcsr.step on completed instruction + haltreq + trigger before next instruction
This is the really problematic case. Section 4.4.1 indicates that dcsr.step has priority over trigger. The priority listing indicates that trigger has priority over haltreq, and haltreq has priority over dcsr.step. This is what I have posted the issue in the hope of resolving.

Silabs-ArjanB, 2021-12-08:
Hi @JamesKenneyImperas Thank you for clarifying that you were talking about the 'timing' bit. I re-read 4.4.1 including the 'Similarly, a trigger at the new address does not fire until the hart actually attempts to execute that instruction.' statement and I don't see how that section or any statement in it can be interpreted as 'step having priority over trigger.

The part about 'attempts to execute that instruction' relates to a different topic really.

JamesKenneyImperas, 2021-12-08:
Hello @Silabs-ArjanB,

Perhaps this can be clarified in your case if, for each of the three cases I stated, you say which exception cause you would expect when the debug exception is taken.

Silabs-ArjanB, 2021-12-08:
Hi @JamesKenneyImperas 

I would expect the following:

>     dcsr.step on completed instruction + trigger before next instruction
>     In this case, the specification section 4.4.1 implies that step exception must be taken (cause 4)

As explained in the 'cause' bitfield part of the 'dcsr' table, 'step' has lower priority than 'trigger match with action = 1'. I also wonder what you mean by 'trigger before next instruction'; that is not really 'debug spec terminology'. If you mean 'trigger with timing 0', then that seems to conflict with 'completed instruction', if you mean 'timing = 1', then the answer is still that 'step has lower priority than trigger match'.

>     haltreq + trigger before next instruction
>     In this case, priorities indicate that the higher-priority trigger must be taken (cause 2)

haltreq has lower priority than trigger match with action = 1 (per the same table as above). Again the 'timing' bit is not relevant to this discussion, so the answer is the same for timing = 0 or 1.

> dcsr.step on completed instruction + haltreq + trigger before next instruction
>     This is the really problematic case. Section 4.4.1 indicates that dcsr.step has priority over trigger. The priority [...]

Trigger match has the highest priority of the three reasons mentioned.

JamesKenneyImperas, 2021-12-08:
Hello @Silabs-ArjanB,

I'll try to explain what I mean in different terms: I'm sorry if it appears I'm labouring the point here, but I do believe there is an inconsistency in the specification that needs nailing.

1. Consider the case where we are single-stepping instruction _A_ and the next instruction to execute in program order after that would be instruction _B_ (this is what I mean by _current_ and _next_ instruction - nothing to do with trigger timing). If only `dcsr.step` was active, then we would return from Debug mode, execute instruction _A_, then re-enter Debug mode with cause 4. This is clear.

2. Now suppose that there is an address trigger programmed to trigger on the address of instruction _B_ before it executes and we repeat the experiment of single-stepping instruction _A_. In this case, we should get the same behavior, because the specification says in 4.4.1 that, if a single-step has just occurred, _a trigger at the new address does not fire until the hart actually attempts to execute that instruction_. So, again, we would re-enter Debug mode with cause 4 after executing instruction _A_.

3. Now suppose that we repeat the experiment again (single step _A_ + address trigger on _B_) but raise `haltreq` during the execution of instruction _A_. I think you are expecting that debug mode would be entered after execution of _A_ with cause 3 (haltreq) - correct? Note that it can't be the address trigger on instruction _B_ that is reported, because there has been no attempt to execute that instruction, just as in case 2.

4. Finally, suppose that we repeat the experiment once more, raising `haltreq` during the execution of instruction _A_ and with the address trigger on the address of instruction _B_, but with `dcsr.step=0`. After executing instruction _A_, we have both `haltreq` pending and a higher-priority address trigger pending on the next instruction address (_B_). Here, I think you say that you would expect the address trigger (cause 2) to be reported because it is higher priority.

Cases 3 and 4 are not consistent, because although in both cases `haltreq` and address trigger are pending the first case reports cause 3 and the second case cause 2. The exact cause reported appears to be affected by whether `dcsr.step` is set.

Apologies for the long treatise!

silabs-oysteink, 2021-12-08:
Here are my two cents on this issue:

Case 1: I agree, A is executed, debug mode with dpc pointing to B is entered and B will be executed after we return from debug mode.

Case 2:  I agree that this should have the same behavior as for case 1. If we stretch the case a bit, when instruction B is attempted executed, it will cause a trigger match and enter debug mode, _without_ instruction B executing. DPC will once again point to instruction B.

Case 3: For our pipeline the outcome of this scenario depends on what type of instruction is being executed and how far it has progressed in the pipeline. Any non-load/store instruction can be interrupted at any time, so if a haltreq occurs the core will simply kill A, and jump to debug with dpc pointing to A. If however, A is either a load or store, and is far enough down the pipeline to have started it's bus transaction, we cannot kill it and have to let it finish. At the time it finishes, we have a pending single step AND a pending haltreq. According to the debug spec haltreq has higher priority and has to be reported in dcsr.cause, dpc will point at B.

Case 4: This again depends on the type of instruction and progress in the pipeline as for case 3. If we are allowed to kill A, we will do that and enter debug mode with cause=3 and dpc pointing at A. After we exit debug mode, A will execute, then B will be attempted executed but will fire a trigger match and enter debug with dcsr.cause=4 and dpc point at B.

If we are not allowed to kill A, A will finish and then B will be attempted executed. It will cause a trigger match and enter debug with cause=4 and dpc pointing at B. Note that this also depends on the timing relationship between A and B. If there are any pipeline bubbles after A in this case, we will enter debug with dcsr.cause=3 and dpc pointing at B.

I am not sure if this made it more clear, or if I just opened a new can of worms though.

Silabs-ArjanB, 2021-12-08:
Hi @JamesKenneyImperas Agreed with 1, 2, and 3. Also agreed with the detailed explanations 1,2,3,4 of Oystein. I think the description of your scenario 4 is conflicting with itself though.

> Finally, suppose that we repeat the experiment once more, raising haltreq during the execution of instruction A and with the address trigger on the address of instruction B, but with dcsr.step=0. [...]

If haltreq is raised during the execution of instruction A, then you cannot assume that 'we have both haltreq pending and a higher-priority address trigger pending on the next instruction address (B).' If this happens, then we agree that the address trigger has higher priority than the haltreq. It is however also possible that instruction B is not yet attempted to execute (it might even not have been fetched for all we know), and in this case haltreq is the only reason to enter debug and therefore debug will be entered just because of that reason (there is no 'must/requirement' that the CPU should try to execute a next instruction and see what happens before it is allowed to have a look at haltreq).

JamesKenneyImperas, 2021-12-08:
Hi @Silabs-ArjanB and @silabs-oysteink,

I think we can ignore cases where in-flight instructions are killed. From the Imperas simulator / cosimulation verification perspective, this is as if the instruction had not started at all. In the case that you initially reported, I think the instruction being single-stepped is a load and it therefore can't be killed, correct?

Assuming non-killable instructions for case 3, you say Oystein that the haltreq is taken in preference to the step breakpoint. (Section 4.4.1 forbids the trigger match on the next instruction being reported at this point, so it definitely can't be that.)

Assuming non-killable instructions for case 4, it sounds by combining your answers that either haltreq _or_ trigger match might be reported, depending on whether there are pipeline bubbles or other things that the Imperas simulator does not model -is this correct? _Note that this differs from case 3 where trigger match is expressly forbidden_.

I think we can update the Imperas reference so that in the special case that both step breakpoint _and_ haltreq are pending on completion of an instruction then haltreq is reported, which will probably make the test you provided pass, although the logic is odd. But is there a more general problem that needs addressing with respect to haltreq and trigger match? Does the comparison need to be somewhat flexible here to allow for the case that either might be taken if both are possible at the point an instruction starts?

(I'm not sure whether Lee is already doing anything to handle this kind of thing in the harness - I'll check with him tomorrow.)

Silabs-ArjanB, 2021-12-09:
Hi @JamesKenneyImperas,

> 
I think we can ignore cases where in-flight instructions are killed. From the Imperas simulator / cosimulation verification perspective, this is as if the instruction had not started at all. In the case that you initially reported, I think the instruction being single-stepped is a load and it therefore can't be killed, correct?

Yes correct (if an instruction, no matter whether it is single stepped or not, already had visible side effects, like for example a load or store going on the data bus, then it will never get killed).

>> Assuming non-killable instructions for case 3, you say Oystein that the haltreq is taken in preference to the step breakpoint. (Section 4.4.1 forbids the trigger match on the next instruction being reported at this point, so it definitely can't be that.)

Correct (we would not even fetch instruction B for various reasons, one of them being that if B would also be non-killable, then we could end up in the situation where a 'single' step leads to execution of more than 1 instruction.).

> Assuming non-killable instructions for case 4, it sounds by combining your answers that either haltreq or trigger match might be reported, depending on whether there are pipeline bubbles or other [...]

Correct. If A is non-killable then the debug cause will depend on whether there is a bubble between A and B. If there is a bubble in the pipeline after A, then the debug request 'wins' (B gets killed, the PC of B will be in dpc). If there is no bubble after A, then B's trigger match will be seen and will 'win' against the debug request. Indeed, there is no way for a non-cycle-accurate ISS to know what will happen.

>> I think we can update the Imperas reference so that in the special case that both step breakpoint and haltreq are pending on completion of an instruction then haltreq is reported, which will probably make the test you provided pass, although the logic is odd. 

Okay (but I don't see why you consider this a 'special case' as the debug spec states "when there are multiple reasons to enter Debug Mode in a single cycle, hardware should set cause to the cause with the highest priority").

>>But is there a more general problem that needs addressing with respect to haltreq and trigger match? 

It is true that after starting (i.e. fetching) an instruction that would 'hit' a trigger match and there is a haltreq at some point when this instruction is inside the pipeline, then the RTL might either cause the trigger to 'fire' and cause debug entry because of the trigger match or the haltreq might win and would be the cause of debug entry. 

We do expect this is already working correctly.

>>Does the comparison need to be somewhat flexible here to allow for the case that either might be taken if both are possible at the point an instruction starts?

We do not think more flexibility is needed here in the ISS as in the scenario that haltreq wins, then RVFI would not even signal anything about instruction B.

>> (I'm not sure whether Lee is already doing anything to handle this kind of thing in the harness - I'll check with him tomorrow.) 

So we do not think a change is required related to trigger match, but we do still think that "the ISS should not assume that debug is entered due to single step even if a single stepped instruction completed" as only on the next instruction after that can the ISS know what the highest priority debug entry reason was.

We do agree that the debug specification is a bit ambiguous because it combines asynchronous and synchronous reasons in the same priority list and we agree with where the discussion in https://lists.riscv.org/g/tech-debug/topic/86584183 seems to be going (although the debug spec does not reflect that discussion). We believe the RTL is inline with the intent here at least.

eroom1966, 2021-12-09:
Pull request issued, could somebody please review and comment

Silabs-ArjanB, 2021-12-10:
Should be fixed with https://github.com/openhwgroup/core-v-verif/pull/1062.

@strichmo Did this fix work and can this be closed now?

silabs-robin, 2021-12-22:
> Did this fix work and can this be closed now?

I have tested whether the fix in the ISS has resolved this issue.
Before adding the new ".so", the given hash/seed reproduces the issue.
After adding the new ".so", the same pre-conditions are reproduced (observed in trace) but the ISS gives a "dcsr.cause" that agrees with the RTL, and the test runs succesfully til completion.

From that experiment, it looks to me like the fix works and that the issue can be closed.
@Silabs-ArjanB and @silabs-oysteink, you can see if you agree that the following trace shows the same scenario as originally posted, but with the ISS behaving as we would expect.

[image]

Imperas, 2021-12-22:
I think this should be closed, Steve Richmond already merged the pull request as far as I can recall

Silabs-ArjanB, 2022-01-06:
@silabs-oysteink Can you check the question asked by Robin above and if 'all okay' close this issue?

silabs-oysteink, 2022-01-06:
It seems to be all okay, closing the issue.

closed 2022-01-06 by silabs-oysteink