openhwfoundation/cv32e40x #342: rvfi_trap not correct for single step instructions

silabs-oysteink, 2021-12-06:
## Bug Title
rvfi_trap not correct for single step instructions

### Component
**Component:RTL**: RVFI

### Steps to Reproduce
The steps provided in issue #340 can be used for this issue as well. Any single stepped instruction in the test seems to exhibit this behavior:

The single step instruction reaches WB and generates a wb_valid, which leads to rvfi_valid with no bits set in rvfi_trap. rvfi_trap_next can be seen to have some bits set at the same time as rvfi_valid is high (marked in blue in the figure), but these does not seem to correspond to the user manual description of rvfi_trap.

[image]
.

silabs-halfdan, 2021-12-28:
RVFI uses the pc_mux to detect synchronous debug entries. As the single step debug entry updates the mux the cycle after wb_valid, it will need to be triggered separately and can not use the exception code signal from ctrl_fsm_o.
[image]
.

Silabs-ArjanB, 2022-01-04:
@silabs-oysteink Can this one be closed now?

silabs-oysteink, 2022-01-04:
Yes, closing it.

closed 2022-01-04 by silabs-oysteink