openhwfoundation/cv32e40x #345: Naming and timing of debug related signals

Silabs-ArjanB, 2021-12-09:
The current RTL uses pending_single_step to signal a pending single step and it uses pending_debug for all other debug reasons. Also the pending_debug signal factors into the pending_single_step signal (see https://github.com/openhwgroup/cv32e40x/blob/10763a517a4dbdbccf2ce050273177cb4a5b7bd9/rtl/cv32e40x_controller_fsm.sv#L294) which is done to make the pending single step the lowest priority.

The following should be considered:

- Don't mix the pending signals with priority logic, so remove  pending_debug from the pending_single_step expression.
- Rename debug_pending and debug_allowed to make it clear that these exclude single step.
- Could pending_single_step be raised a cycle later (to avoid the dependency on wb_valid_i and to make the timing for debug entry after a load/store the same for single stepping or external debug request)?
- Rewrite the debug_cause_n assignment so that it is ordered according to priority.

The current logic assumes that a pending single step is either taken immediately (if there is no higher priority debug entry reason) or never (when there is a higher priority reason). This works currently because the external debug request cannot be withdrawn, but if we change that (e.g. because of https://github.com/openhwgroup/cv32e40x/issues/341), then that might lead to the need to introduce a sticky version of the pending_single_step signal.

silabs-oysteink, 2022-12-08:
- PR #719 removes other pending_debug causes from the pending_single_step signal.
- The pending_debug and debug allowed are already split into pending_async/async_allowed and pending_sync/sync/allowed. Step is not included/excluded in the naming though.
- debug_cause_n assignment updated for all causes with priority accoring to debug spec 0.13.2 (update to 1.0.0 will change the priority)

- No update for delaying the single step entry. Should consider if we should delay it, and if delaying it also implies we should delay trigger matches for instance.

Silabs-ArjanB, 2022-12-09:
Sufficiently addressed with comments added in https://github.com/openhwgroup/cv32e40x/pull/721

closed 2022-12-09 by Silabs-ArjanB