openhwfoundation/cv32e40x #347: rvfi reports mcountinhibit write incorrectly

strichmo, 2021-12-10:
### Component
**Component:Other**: RVFI issue

### Steps to Reproduce
1. Use this fork: https://github.com/strichmo/core-v-verif.git
2. Use the strichmo/pr/mcountinhibit fork, hash: 1c38c07d7c80fd35a6ec20420093b0deffe599d3
3. Run a performance counter test
```
% makeuvmt test TEST=perf_counters_instructions
```

The test performs a write of all ones (0xffff_ffff) to mcountinhibit.  Upon this write the testbench should flag an error:
```
UVM_ERROR @ 1749.300 ns : uvme_cv32e40x_core_sb.sv(381) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 141, pc: 0x000000fe, csr: mcountinhibit, rvfi = 0xffffffff, rvvi = 0x0000000d, mask = 0xffffffff
```

The reason for this is that the RVFI reports a write mask and write data of all ones for this instruction, implying that the mcountinhibit value should be 0xffff_ffff, but it should be 0xd in this configuration.

**Note: Please merge the above branch into core-v-verif (cv32e40x/dev) upon fixing this bug in RVFI**

Silabs-ArjanB, 2022-01-04:
Fixed with https://github.com/openhwgroup/cv32e40x/pull/364

@silabs-hfegran Could you please perform the merge that Steve requested above in core-v-verif (after which this issue can be closed)?

silabs-hfegran, 2022-01-04:
Branch merged, this issue should be resolved

closed 2022-01-04 by silabs-hfegran