openhwfoundation/cv32e40x #386: mcause mismatch when writing to mstatush

silabs-robin, 2022-01-11:
### Component
**Component:RTL**
**Component:Other**

### Steps to Reproduce
Latest core-v-verif (e85dd3).
Run `csrrw x7, 0x310, x31` in any test.
For example, modify `hello-world` to run this:
```
int main(void)
{
    printf("TODO hello\n");
    
    __asm__ volatile("nop");
    __asm__ volatile("nop");
    __asm__ volatile("nop");
    __asm__ volatile("csrrw x7, 0x310, x31");
    __asm__ volatile("nop");
    __asm__ volatile("nop");
    __asm__ volatile("nop");
    
    printf("TODO goodbye\n");

    return EXIT_SUCCESS;
}
```

The first error message is:
`CSR Mismatch, order: 3185, pc: 0x000000d8, csr: mcause, rvfi = 0x00000000, rvvi = 0x00000002, mask = 0xffffffff`

It looks like the ISS is flagging this CSRRW as an illegal instruction?
But according to riscv spec and 40x user manual, CSR 0x310 `mstatush` should be RMW.
I think the core is doing the right thing but I would like a second opinion on it.

[image]

The issue was first found in `corev_rand_debug_ebreak` test at hash `3951c0` and seed `2048991240`.

silabs-hfegran, 2022-01-12:
mstatush was initially defined in privilege spec v1.12, which the ISS does not have enabled - I am currently working on enabling that through the common configuration object in core-v-verif

silabs-robin, 2022-01-12:
Thanks @silabs-hfegran and @Silabs-ArjanB for pointing out that "mstatush" only is in priv spec v1.12.
I did some tests and the problem goes away when configuring the ISS to use `--override root/cpu/priv_version=master`.
Once @silabs-hfegran has done the v1.12 update this will be fixed.

closed 2022-01-12 by silabs-robin