openhwfoundation/cv32e40x #398: Assert a_nmi_handler_max_retire fails in formal with PMA config

silabs-mateilga, 2022-01-18:
**Component:RTL**
git hash: 9c1d3db443c862d41ddf17dc0ba97ace46e78aa3

Controller retires two instructions (1 is max) before taking the bus error NMI, when an ST instruction is retired to the write buffer. See waveform0. for details.

Early debugging with @silabs-oysteink indicates id stage should be halted in the case where nmi_allowed is high but the fsm decides not to take it anyway. 
A quick modification where nmi_allowed is made to depend on nmi_pending_q (   assign nmi_allowed = interrupt_allowed && nmi_pending_q;   ) solved the immediate CEX, but formal found another CEX seen in waveform1.

reproduce: Run formal verification with PMA enabled, any of the PMA configurations found in core-v-verif/cv32e40x/tb/uvmt/uvmt_cv32e40x_constants.sv should show the problem, example waveform from pma_test_cfg_4.

waveform0
[image]

waveform1
[image]

Silabs-ArjanB, 2022-02-21:
@silabs-mateilga Can this be closed now?

silabs-mateilga, 2022-02-21:
@Silabs-ArjanB yes, closing it now.

closed 2022-02-21 by silabs-mateilga