openhwfoundation/cv32e40x #402: Issue illegal instruction for deconfigured B extensions

silabs-oivind, 2022-01-21:
Illegal instruction won't be issued from the B decoder if trying to execute a deconfigured instruction.

In the example below (and other similar ones), add "else  decoder_ctrl_o = DECODER_CTRL_ILLEGAL_INSN":

         {7'b001_0000, 3'b010}: begin // Shift left by 1 and add (sh1add)
            if (RV32B_ZBA) begin
              decoder_ctrl_o.alu_operator = ALU_B_SH1ADD;
            end
          end

Silabs-ArjanB, 2022-01-24:
Also put the carryless multiplication circuit in the ALU in a generate statement as it it not obvious that it will get optimized away if not used.

Silabs-ArjanB, 2022-12-15:
@silabs-oivind  Please close this issue if you agree it has been solved.

closed 2022-12-15 by silabs-oivind