openhwfoundation/cv32e40x #404: User interrupt enable present in mstatus bitfields description

silabs-hfegran, 2022-01-24:
## User interrupt enable present in mstatus bitfields description

### Steps to Reproduce
Last line in the section below:
[image]

silabs-hfegran, 2022-01-24:
There is also an overlap of MPP and unimplemented bit 12

Silabs-ArjanB, 2022-01-24:
Hi @silabs-hfegran I don't understand the issue. The RTL and the documentation look fine except that the 40X documentation should not even mention user mode and can just state that bits 12:11 are RO (0x3) (which it basically says, but it says it referring to user mode). 

For the 40S we should (internally) document the implemented WARL behavior of this field.

silabs-hfegran, 2022-01-24:
Hi @Silabs-ArjanB, 
The UM is contradicting itself, bit 12 should not be included in the RO hardwired 0-field, it should be 
15:13 = hardwired 3'h000 (as opposed to 15:_12_ )
12:11 hardwired 2'h11.

Silabs-ArjanB, 2022-01-24:
@silabs-hfegran Now I see, agreed. Thanks.

Silabs-ArjanB, 2022-02-21:
Hi @silabs-hfegran This should be fixed now in:

https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/latest/control_status_registers.html#machine-status-mstatus
https://docs.openhwgroup.org/projects/cv32e40x-user-manual/en/latest/control_status_registers.html#machine-status-mstatus