openhwfoundation/cv32e40x #406: RVFI Tracer supports out-of-date `scontext` CSR and not `mscontext`.

MikeOpenHWGroup, 2022-01-24:
### Component
`./bhv/cv32e40x_rvfi.sv` and  `./bhv/include/cv32e40x_rvfi_pkg.sv` reference the **scontext** CSR.  This is from v0.13.2 of the RISC-V External Debug Support specification and has been supplanted by **mscontext**.

### Steps to Reproduce
Visual inspection

### Additional Information
This is a low priority issue (for now, anyway).  As far as I can tell, the RTL does not yet support **mscontext**, and the RVFI Tracer merely sets these to idle.

Silabs-ArjanB, 2022-01-25:
There are more CSRs missing that are already in the user manual. Let's first add the couple of remaining to be implemented CSRs in the RTL and then update the RVFI adapter to be complete for both the 40X and 40S.

Silabs-ArjanB, 2022-02-21:
Rename scontext -> mscontext in following files:

- bhv/include/cv32e40s_wrapper.vh 
- bhv/cv32e40s_rvfi.sv
- bhv/include/cv32e40s_rvfi_pkg.sv

silabs-oysteink, 2023-05-08:
This issue seems obsolete as the *context CSRs are not implemented and removed from RVFI,  RTL and user manual.
Do you agree @MikeOpenHWGroup ?

MikeOpenHWGroup, 2023-05-08:
Agreed.  Closing the issue.  Thanks!

closed 2023-05-08 by MikeOpenHWGroup