openhwfoundation/cv32e40x #408: Use of c++ keywords in SystemVerilog

Nicolas-Gaudin, 2022-01-25:
## Use of c++ keywords in SystemVerilog

The CV32E40X systemverilog uses **float** and **interrupt** as variable names.

I get SYMRSVDWORD warnings on struct declarations in if_xif and and pkg components when I compile the core with verilator under a core-v-verif environment. The issue comes from verilator and its use of c++ to compile the core. 

### Component
2 warnings in if_xif.sv : (float)
https://github.com/openhwgroup/cv32e40x/blob/d25e3b5042c4d993271149c0cce5d12a88861018/rtl/if_xif.sv#L66
https://github.com/openhwgroup/cv32e40x/blob/d25e3b5042c4d993271149c0cce5d12a88861018/rtl/if_xif.sv#L105

1 warning in cv32e40x_pkg.sv : (interrupt)
https://github.com/openhwgroup/cv32e40x/blob/d25e3b5042c4d993271149c0cce5d12a88861018/rtl/include/cv32e40x_pkg.sv#L527

### Steps to Reproduce
//Having already Verilator and the riscv TC
> git clone https://github.com/openhwgroup/core-v-verif
> cd core-v-verif
> git checkout cv32e40x/dev
> cd cv32e40x/sim/core
> #% set environment variables 
> #% CV_SW_TOOLCHAIN = /opt/riscv
> #% CV_SW_PREFIX = riscv32-unknown-elf-
> make
> #% make command will compile the core using verilator and then simulate the core with a 'hello-world' code

Result:
```console
%Warning-SYMRSVDWORD: /opt/core-v-verif/core-v-cores/cv32e40x/rtl/include/cv32e40x_pkg.sv:527:19: Symbol matches C++ common word: 'interrupt'
  527 |   logic           interrupt;
      |                   ^~~~~~~~~
%Warning-SYMRSVDWORD: /opt/core-v-verif/core-v-cores/cv32e40x/rtl/if_xif.sv:66:11: Symbol matches C++ keyword: 'float'
   66 |     logic float;       
      |           ^~~~~
%Warning-SYMRSVDWORD: /opt/core-v-verif/core-v-cores/cv32e40x/rtl/if_xif.sv:105:32: Symbol matches C++ keyword: 'float'
  105 |     logic                      float;    
      |                                ^~~~~
```

Silabs-ArjanB, 2022-01-25:
Let's change the following:

- interrupt -> irq in mcause_t
- Remove float from x_issue_resp_t and in x_result_t (they were already removed from https://github.com/openhwgroup/core-v-xif)

And because we will touch  cv32e40x/rtl/if_xif.sv anyway:

- Add dbg to x_mem_resp_t and x_mem_result_t as per https://docs.openhwgroup.org/projects/openhw-group-core-v-xif/ (and assign properly inside core)

Silabs-ArjanB, 2022-02-02:
Hi @Nicolas-Gaudin , this issue should have been fixed now. Thank you for reporting it. Please close it if you agree it has been fixed.

closed 2022-02-02 by Nicolas-Gaudin