openhwfoundation/cv32e40x #453: "mstatus.UIE" shouldn't exist

silabs-robin, 2022-02-22:
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

The RTL has the field [mstatus.uie](https://github.com/openhwgroup/cv32e40x/blob/master/rtl/include/cv32e40x_pkg.sv#L497), but in the riscv priv spec there is no such thing.

Suggested fix: `logic [2:0]   zero0`

Silabs-ArjanB, 2022-02-23:
Hi @silabs-robin I created https://github.com/openhwgroup/cv32e40s/pull/146 and https://github.com/openhwgroup/cv32e40x/pull/455 to remove this dead code (both are SEC clean as these bitfields were not used anywhere)

Silabs-ArjanB, 2022-02-23:
Hi @silabs-robin Please close this issue if you agree it has been addressed properly.

silabs-robin, 2022-02-23:
Jepp. I was waiting for the merge.
The issue has been addressed properly.

closed 2022-02-23 by silabs-robin