openhwfoundation/cv32e40x #454: mimpid implementation no longer hardcoded

MikeOpenHWGroup, 2022-02-22:
As of `320925c2` the value of CSR `mimpid` is set by a primary input to the core.  This means that any instantiation of the CV32E40X can set `mimpid` to any value.  This is incompatible with the [core versioning scheme](https://cv32e40x-user-manual.readthedocs.io/en/latest/core_versions.html#core-versions-and-rtl-freeze-rules) developed for CV32E40P.

Is there a reason for this change?  If so, how will the versioning rules of the core be maintained?

davideschiavone, 2022-02-23:
Agree with @MikeOpenHWGroup - dangerous to set it as input

Silabs-ArjanB, 2022-02-28:
Hi @MikeOpenHWGroup , we need part of mimpid to be easily settable via a metal only change and that is why we made this change. I agree that we should not have made the entire 32 bit user definable and therefore that part is undone in https://github.com/openhwgroup/cv32e40x/pull/460. This PR also better documents the intended use of mimpid.

Silabs-ArjanB, 2022-03-03:
mimpid_patch_i is a pin as opposed to a parameter. the documentation states 'The PATCH bitfield is defined and set by the integrator and shall be set to 0 when no patches are applied.' I don't think this is something we can/should check with assertions.

davideschiavone, 2022-03-03:
the assertion IMO should be related to the frozen tag, like if you freeze version v1.0.0, then that pin must be 0, right? that's why I am suggesting that for the frozen version we add this assertion - it doesn't have to kill the simulation, but just reporting it would be fine such that we are sure the user is aware of

MikeOpenHWGroup, 2022-03-03:
Thanks @Silabs-ArjanB and @davideschiavone.  This is a good solution.  My original query has been answered and #460 addresses the implementation so I will close this issue.

closed 2022-03-03 by MikeOpenHWGroup