openhwfoundation/cv32e40x #458: Take MRET debug exception from WB

Silabs-ArjanB, 2022-02-25:
In https://github.com/openhwgroup/cv32e40x/pull/456 an RVFI fix was done that likely could better have been done in the RTL instead.

Fix that should be undone in RVFI:

assign pc_mux_exception = (ctrl_fsm_i.pc_mux == PC_TRAP_EXC) || (ctrl_fsm_i.pc_mux == PC_TRAP_DBE);

vs.

assign pc_mux_exception       = (ctrl_fsm_i.pc_mux == PC_TRAP_EXC) || pc_mux_debug_exception ;
  assign pc_mux_debug_exception = (ctrl_fsm_i.pc_mux == PC_TRAP_DBE) && !dret_in_ex_i; // Ignore exceptions from instructons that will never be executed
  
In the RTL we should consider taking the MRET related debug exception from WB instead (as we do for other exceptions). Check potential impact on PC hardening logic as well.

Silabs-ArjanB, 2022-02-25:
The following code should be updated as well:

assign ctrl_fsm_o.jump_in_id_raw = (alu_jmp_id_i && alu_en_id_i) || (sys_mret_id_i && sys_en_id_i && !debug_mode_q);

Silabs-ArjanB, 2022-06-03:
Some further updates are needed here

silabs-oysteink, 2023-04-26:
I believe this should be fixed now with the merge of PR #442 on cv32e40s.

closed 2023-04-26 by Silabs-ArjanB