openhwfoundation/cv32e40x #466: RVFI needs to expose wakeup irq separate from mip

silabs-mateilga, 2022-03-01:
We need a new signal in the RVFI which looks at the irq_wu_ctrl_o signal and captures the causing state in the mip, and exposes this at the next instruction retire.

[image]

### Background information
This is required to show why the core has woken from a wfi instruction when the interrupt isn't globally enabled. This information is lost from the mip at next instruction retire when the interrupt signal causing a wake up changes (i.e. is "reset" to 0) before the next instruction retires. This is necessary context for the ISS.

### Completion Criteria
A new signal is implemented showing the required information

silabs-mateilga, 2022-03-01:
This task is spawned from the following ticket: [core-v-verif 1193](https://github.com/openhwgroup/core-v-verif/issues/1193)

silabs-robin, 2022-03-01:
Could this be signaled via rvfi's csr `_rdata` for mip? Or do we need a new rvfi signal?

Silabs-ArjanB, 2022-03-01:
We need a new signal. We need the original _rdata signal as that would for example be used if the first instruction after the WFI is a CSR read of MIP. Also we need the original signal for the case that an IRQ is actually taken.

Silabs-ArjanB, 2022-03-08:
The following signals will be added to RVFI:

- rvfi_wu.wu
- rvfi_wu.interrupt
- rvfi_wu.debug
- rvfi_wu.cause
- rvfi_sleep

closed 2022-06-21 by silabs-mateilga