openhwfoundation/cv32e40x #494: Update RVFI to correctly handle "in_trap" for vectored CLIC interrupts

silabs-oysteink, 2022-04-06:
For vectored CLIC interrupts, the following events happen:

1. IRQ becomes pending
2. IRQ is taken, irq_ack=1
3. Pointer is fetched
4. A jump is performed to the loaded pointer
5. First instruction of handler arrives in IF

It is easy to adapt the RVFI in_trap logic to detect the first instruction of the handler, but the assertion "a_irq_ack_rvfi_capture" needs a bit more work to handle the larger delay between irq_ack and the first handler instruction.

Silabs-ArjanB, 2022-09-16:
@silabs-oysteink Has this issue been fixed now (and can it be closed)?

silabs-oysteink, 2022-09-16:
I thought it was, as the (design) rvfi assertions pass. However, @silabs-hfegran  reported issue #300 on the E40S that indicates the issue is still present. A possible solution there would be to let the clic pointers use the first/last_op flags as regular instructions.

silabs-oysteink, 2023-03-16:
Issue #300 on cv32e40s is closed, and the assertion mentioned in this issue is passing. Closing issue.

closed 2023-03-16 by silabs-oysteink