openhwfoundation/cv32e40x #498: Implement proper mret when mcause.minhv==1

silabs-oysteink, 2022-04-06:
If mcause.minhv is set when we execute an mret, we shall not restore the pc to the contents of mepc, but instead do a pointer fetch M[mepc] as for the CLIC pointer fetch as a data access with lsu privilege.

To achieve this, the pipeline must be stalled until EX and WB are empty before starting the pointer fetch. This is because any instructions in EX  or WB may still cause exceptions, and the pointer fetch would use the state POINTER_FETCH in the controller - which does not handle exceptions.

Silabs-ArjanB, 2022-09-17:
Our interpretation of the CLIC specin above issue description is not correct. The xinhv bit of the previous privilege mode should be used. Re-reading the 0.9-draft spec the spec is actually clear about it. The latest CLIC spec has pseudo code for the _x_ret which clarifies it even better.

See also the following:

https://github.com/riscv/riscv-fast-interrupt/issues/272
https://github.com/riscv/riscv-fast-interrupt/pull/273

Silabs-ArjanB, 2022-09-20:
Please have a look at the comments I left in https://github.com/openhwgroup/cv32e40x/pull/660. It seems there are three bugs:

- Not using inhv bsaed on previous privilege
- Mixing ID stage and WB stage timing signals
- mret impacting inhv bit (I think that is not in the actual CLIC spec; please let me know if it is in the spec)

silabs-oysteink, 2022-11-15:
This should now be resolved