openhwfoundation/cv32e40x #501: Implement WARL bitfields in standard manner

Silabs-ArjanB, 2022-04-06:
Following does not follow our WARL spec:

-  mtvec_n.mode             = csr_mtvec_init_i ? mtvec_q.mode : {1'b0, csr_wdata_int[0]};

Need to define uniform approach for implementing WARL behavior that allows for easy diffing between 40S and 40X.

This also addresses an open item from https://github.com/openhwgroup/cv32e40x/pull/492 about about mtvec_n.mode line.

Silabs-ArjanB, 2022-07-27:
Proposing to add WARL functions in _pkg file (if they differ for 40X and 40S) (preferred) or maybe (less preferred) in cs_registers (if they exist for only one core or are identical)

See:

  function automatic privlvl_t dcsr_prv_resolve
  function automatic logic [1:0] mstatus_mpp_resolve

silabs-oysteink, 2023-05-08:
I believe the suggested fix with resolution functions have already been implemented on the latest RTL.

closed 2023-05-08 by Silabs-ArjanB