openhwfoundation/cv32e40x #549: mcause reset value 0x0000_0000 with SMCLIC=1 violates CLIC spec

silabs-hfegran, 2022-05-23:
## mcause reset value 0x0000_0000 with SMCLIC=1 violates CLIC spec

> 5.7. Changes to xcause CSRs 
(...) The mcause.mpp and mcause.mpie fields mirror the mstatus.mpp and mstatus.mpie fields, and are aliased into mcause to reduce context save/restore code.

mcause
[image]

mstatus
[image]

Currently mcause resets to 0x0000_0000 with SMCLIC parameter = 1, this causes a mismatch between the hard-coded 2'b11 value of mstatus.mpp and mcause.mpp after reset. Furthermore, 2'b00 is not a supported value with the WARL-scheme for this register. 

### Component
**Component:RTL**: CSR behavior after reset
**Component:Doc**: User manual CSR description

### Steps to Reproduce
Currently only on internal development branch, problem is present as soon as CLIC is enabled in RTL and ISS.

Silabs-ArjanB, 2022-05-24:
Hi @silabs-hfegran I think (by luck) the RTL is actually correct, but it is quite confusing and not written in our regular way, which likely caused it to be hooked up wrongly to RVFI (which I guess is what you then observed). I will clean up the RTL and what values RVFI gets to see.

silabs-hfegran, 2022-05-24:
Great, thanks for looking into this @Silabs-ArjanB

Silabs-ArjanB, 2022-05-30:
@silabs-hfegran This should be fixed now.

PR #555 (2022-05-27): Fix for issue #549. Clean up CS registers syntax. Tie RVFI to RTL ins…
…tead of local tie offs

SEC clean for SMCLIC = 0.
SEC clean for SMCLIC = 1.
Design assertions pass.

Fixes #549
closed 2022-05-30 by silabs-hfegran