openhwfoundation/cv32e40x #558: mret during debug mode is incrementing minstret

silabs-oysteink, 2022-05-31:
The E40P, E40X and E40S all jump to the debug exception handler if an mret is executed while in debug mode. The debug spec specifies 'undefined' behavior of this scenario.

For the X and S (I haven't checked E40P), this means that the instruction is not marked as illegal, and thus will increment minstret when it completes. In addition, the jump to the debug exception handler is performed when the mret is in the ID stage, as opposed to all other exceptions being taken from the WB stage. 

A proposed fix is to mark the mret during debug mode as illegal in the decoder, and let it reach WB before the jump to the debug exception handler is taken. This will also avoid incrementing minstret as it is marked as illegal. Taking all exceptions from WB is more consistent and also simplifies how RVFI currently handles exceptions.

The proposed fix does however generate a mismatch with the ISS due to the E40X no longer incrementing minstret on an mret during debug.

@eroom1966 : It seems like ISS is incrementing minstret for mret during debug, can you confirm this? This proposed RTL change would also need a similar change in the ISS after we update the user manual and make a new revision.

Silabs-ArjanB, 2022-06-03:
RTL fixed.
Now waiting for matching ISS fix.

eroom1966, 2022-06-17:
@silabs-oysteink 

Thanks I reproduced by running the core-v-verif branch cv32e40x/dev
Unfortunately I am now waiting for some merges from @silabs-mateilga in order to handle the driving of the NMI signals
I cannot use the latest model as it is incompatible with the signal interface

Silabs-ArjanB, 2022-06-17:
>> I shall make this the default setting in the next release, does that sound OK ?

Do you mean 0.2.0. or 0.4.0? Either way, making it the default is fine, ideally already for 0.2.0. Please let us know for what version you will make it the default.

>> and can I ask, should this be the setting for all cores E40P/S/X ?

E40S and E40X (nothing changed for the 40P)

closed 2022-08-04 by silabs-oysteink