openhwfoundation/cv32e40x #569: Double import of package in cv32e40x_compressed_decoder.sv

spersvold, 2022-06-05:
## Double import of package in cv32e40x_compressed_decoder.sv
When compiling cv32e40x with Synopsys VCS I noticed the following warning : 

Note-[SV-LCM-PPWI] Package previously wildcard imported
../../../../cv32e40x/rtl/cv32e40x_compressed_decoder.sv, 42
cv32e40x_compressed_decoder
  Package 'cv32e40x_pkg' already wildcard imported. 
  Ignoring cv32e40x_pkg::*
  See the SystemVerilog LRM(1800-2005), section 19.2.1.

### Component
**Component:RTL**

spersvold, 2022-06-05:
This relates to merge request #572

Silabs-ArjanB, 2022-06-06:
@spersvold Thank you for reporting this. Your https://github.com/openhwgroup/cv32e40x/pull/572 PR has been merged now. Thanks for your contribution.

closed 2022-06-06 by spersvold