openhwfoundation/cv32e40x #596: unique case statements consistently triggers warnings/non-fatal errors during reset

silabs-hfegran, 2022-06-28:
```   
 always_comb begin
    unique case ({count_up, count_down})
      2'b00 : begin
        next_cnt = cnt_q;
      end
      2'b01 : begin
        next_cnt = cnt_q - 1'b1;
      end
      2'b10 : begin
        next_cnt = cnt_q + 1'b1;
      end
      2'b11 : begin
        next_cnt = cnt_q;
      end
    endcase
  end 
  ```
  
Code such as the above consistently triggers warnings/non-fatal error at the start of simulation due to no case item matching. 
This clutters logs and generates unnecessary warnings. It would be good if we could rephrase such statements to avoid said warnings, e.g. by using defaults, unique0 or any other appropriate construction. 

Source: 
cv32e40x/rtl/cv32e40x_div.sv:117
cv32e40x/rtl/cv32e40x_load_store_unit.sv:496

Silabs-ArjanB, 2022-06-28:
Hi @silabs-hfegran The use of 'unique' is on purpose and will guarantee us that we handled all cases. A violation here is only possible when the CPU has not been reset yet.

If you have a case where you get similar warnings/errors after reset, can you provide a waveform form that please?

silabs-hfegran, 2022-06-30:
Thanks @Silabs-ArjanB, All cases I have seen appears to be related to reset (at t>0 with rst asserted or t = 0)  - I will close this issue and open a similar one on core-v-verif to investigate possible solutions on the testbench side of things

Silabs-ArjanB, 2022-06-30:
Re-open until the t>0 issue is understood. Might 'fix' the RTL just to get rid of these warnings

silabs-hfegran, 2022-06-30:
For reference, metrics log:
[image]

Silabs-ArjanB, 2022-06-30:
Hi @silabs-hfegran So it seems that reset only becomes asserted at 2 ns, which is why you get warnings at that time as well. Is that being done on purpose?

Silabs-ArjanB, 2022-06-30:
I think it will be easiest if we simply remove the unique keyword in the two reported cases.

silabs-hfegran, 2023-04-13:
[image]
 still seeing the issue here

silabs-oysteink, 2023-05-15:
This should be fixed by PR #452 on cv32e40s.

Silabs-ArjanB, 2023-05-15:
@silabs-hfegran Can you please close this if indeed fixed?