openhwfoundation/cv32e40x #604: E40X 0.4.0 : Doc, Updated jvt and mstateen0 CSRs according to latest Zc* clarifications… by @Silabs-ArjanB in #554

eroom1966, 2022-06-29:
In the Documentation Changes from 0.3.0 to 0.4.0, it says

> Updated jvt and mstateen0 CSRs according to latest Zc* clarifications… by @Silabs-ArjanB in #554

I cannot find any differences in the pdf documentation between 0.3.0 and 0.4.0 referring to mstateen0 , could you please identify the difference in the text.

Imperas, 2022-07-01:
The process for Imperas to update the 40X reference model is to rely on the specification as a frozen pdf, for example version 0.4.0 - and so we do not look at the RTL and we do not look at the documentation 'source' files - we build the model from the specification and so please ensure that the specification pdf is correct (and any changes to the source that creates it is propagated) as that what we rely on and all users will rely on. [After all the verification process if to compare the RTL to the reference model (via the spec)- so it makes no sense for the reference model just to copy potential errors in the RTL...]

Also - the 40X spec should not have items/content in it that relates to other cores (e.g. 40S) - the 40X spec should define the 40X core. period.

So - please fix the pdf.

Also I would recommend the use of incremental changes eg 0.4.0->0.4.1 for changes to the doc that don't change the spec.

Simon

Silabs-ArjanB, 2022-07-01:
@Imperas The changelist is autogenerated and can therefore contain comments that do not actually apply to the specific core itself. Such non-applicable content does not impact (and is not visible in) the actual content (i.e. everything else than the autogenerated change list) of the pdf . If you would simply do a 'pdf diff' on two released versions of the user manual and ignore diffs in the change list itself, you will have an accurate overview of the changes. 

There is not need for Imperas to look in the documentation source files or in the RTL files. Please let me know if this works for you. If not, then we will filter the change list going forward to exclude potentially misleading content.

We can agree to increment the 'patch' part of the version number for incremental changes that don't change the spec, but in reality that will not help you much as user manual releases are infrequent and bundle multiple changes (so the chance of there only being 'patch' releases in such bundle is very low).

Silabs-ArjanB, 2022-10-04:
@eroom1966 Can this ticket be closed now?

closed 2022-10-06 by eroom1966