openhwfoundation/cv32e40x #618: E40X 0.4.0 doc, illegal writes to mcause

eroom1966, 2022-07-13:
mcause.exccode is defined as WLRL

What should happen if an illegal value is written to mcause.exccode ?
- should an exception be taken
- should the value be accepted or previous value retained

Silabs-ArjanB, 2022-07-15:
Hello @eroom1966 ,

exccode[10:0] are implemented as RW bits. The more significant bits are hard-wired to 0, i.e. if Smclic is not included, then exccode[11] = 0, if Smclic is included, the exccode[30:11] = 0.

eroom1966, 2022-07-15:
@Silabs-ArjanB 
This does not answer my question I think, let me ask a specific case

what would happen if I write a value of exccode=4

this is not a legal value for the e40x
Thx
Lee

Silabs-ArjanB, 2022-07-15:
Hi @eroom1966 , if you write exccode = 4 (even if that is an illegal value), then you will read back 4 from that bitfield as exccode[10:0] are implemented as RW bits. Same holds for values 0 .. 2^11 - 1

eroom1966, 2022-07-15:
thanks that is now clear, closing issue.
On a side note, I find it strange that 
WARL **must** return a legal value, yet
WLRL Is **not required** to return a legal value

nonetheless, your intention is clear.

closed 2022-07-15 by eroom1966